atcollab/at #1: ID modelling in AT

willrogers, 2017-08-30:
_From @bnash on July 21, 2017 10:58_

Modeling of ID's is an important area of development for AT.

There are several different ways to model ID's in AT.  The first is simply to create an array of bending magnets.  This has the benefit that the radiation effect is computed correctly, but the field profile is not totally right.  Also, non-linear effects will not be included here.

Secondly, one can use the analytical method of Ying Wu and Forest (others?).  This is implemented in the pass method GWigSymplecticPass.c:
https://github.com/atcollab/at2devel/blob/master/atintegrators/user/GWigSymplecticPass.c

The radiation damping effect may be added in an analagous way as in driftkickrad.c in
https://github.com/atcollab/at2devel/blob/master/atintegrators/BndMPoleSymplectic4RadPass.c
For the diffusion, one would need to modify findmpoleraddiffmatrix.c
https://github.com/atcollab/at2devel/blob/master/atmat/atphysics/findmpoleraddiffmatrix.c
Which is used in OhmiEnvelope.

Finally, one may use the kick map method for tracking through IDs.  This is implemented in 
https://github.com/atcollab/at2devel/blob/master/atintegrators/IdTablePass.c

Here, one also needs to add the radiation effects.  For the damping, one could compute a damping matrix to add to the kick map as a matrix element.  For the diffusion, one could compute the diffusion matrix in Radia along with the kick map.  This matrix would then need to be read in, in a modified
findmpoleraddiffmatrix.c

_Copied from original issue: atcollab/at2devel#2_

lcarver, 2023-10-16:
Between the developments of @oscarxblanco in #558  and the EnergyLoss passmethod in #632 , can this be closed?

oscarxblanco, 2023-10-16:
I believe Simone Liuzzo's work and discussions on issue #571 (Undulator dipoles) should also be cited here for reference.