atcollab/at #5: pyat: Pypi upload

willrogers, 2017-08-30:
_From @willrogers on July 21, 2017 14:23_

_From @willrogers on October 20, 2016 8:44_

I believe we should be able to upload binary 'wheels' to Pypi.

This should allow `pip install pyat`.

I think this will require building with `setuptools` rather than `distutils` (a fairly minor change).

_Copied from original issue: willrogers/at#18_

_Copied from original issue: atcollab/at2devel#6_

willrogers, 2017-08-30:
The ideal situation would be that you could `pip install pyat` from any machine.  There are a number of complications with this:

* the names `pyat`, `atpy` and `python-at` are all taken in pypi.  `at-python` is available
* you can install numpy from wheels, but scipy is not available as a wheel for Windows.  The only need for scipy is to load a lattice from a Matlab file, but this is the primary way we load lattices at present
* building wheels for multiple Linux systems is also a bit tricky.  [This project](https://github.com/pypa/manylinux) uses Docker to enable building the necessary wheels.  [Numpy](https://pypi.python.org/pypi/numpy) uses this approach
  * you can fairly easily build from source on Linux and Mac, but it's not clear to me what the user would need installed to allow this

Building all the wheels to make a release is sounding like a bit of a job.

willrogers, 2017-08-30:
I guess what we should do is upload wheels for Windows and rely on compiling on installation for Linux and Mac.  That still leaves the problem of Scipy on Windows, but I don't see a simple way out of that one.

willrogers, 2017-09-13:
We could use [cibuildwheel](https://github.com/joerick/cibuildwheel) to build the wheels on Appveyor and Travis CI, then publish them to Github Releases.  We could then manually upload them to PyPI.

willrogers, 2017-09-16:
Windows may be able to install numpy and scipy via the packages here: http://www.lfd.uci.edu/~gohlke/pythonlibs/

smartsammler, 2017-11-02:
Good news.
SciPy 1.0 has got wheels for Windows, too, see https://pypi.org/project/scipy/#files 
Though I can't test them the problem with Windows might be solved if one picks SciPy 1.0 which should be ok, since it's only used for ``io.loadmat``.

willrogers, 2017-11-02:
Good news, thanks for the tip.

willrogers, 2020-06-26:
We now have packages on PyPI under the name `accelerator-toolbox`: https://pypi.org/project/accelerator-toolbox/

closed 2020-06-26 by willrogers