atcollab/at #6: pyat: Numpy arrays are transposed with respect to Matlab ones

willrogers, 2017-08-30:
_From @willrogers on July 21, 2017 14:23_

_From @willrogers on March 21, 2017 11:21_

AT requires input of shape `(6, x)`.
pyAT requires input of shape `(x, 6)`.

The difference is confusing.  I cannot remember if there is a technical reason why this is, or if it could be corrected to be the same.

Perhaps @lfarv can remember.

_Copied from original issue: willrogers/at#27_

_Copied from original issue: atcollab/at2devel#7_

willrogers, 2017-08-30:
_From @lfarv on March 21, 2017 12:34_

The integrators, ruling on a set on several particles rely on the fact that the 6 coordinates of one particle are consecutive  in memory, and the 2nd particle follows the 1st one. With the default ordering of Matlab and Python, this corresponds to 
6xN in Matlab and Nx6 in numpy. It should also accept 6xN numpy arrays with Fortran-ordering, but it is probably safer to use the default ordering.

willrogers, 2017-08-30:
Thanks for explaining this.  We have a couple of options here.

* Switch to using Fortran-aligned arrays.  I might prefer this, since the comparability of the output for AT and pyAT improves.  The disadvantage is that the user needs to create Fortran-aligned arrays and they won't be expecting to.
* Keep with C-aligned arrays.  The opposite of the above.
* Don't use `atpass` directly - use a Python function that checks the arguments and converts if necessary.

Usability and absolute speed may conflict here, but I am finding the differences in algorithms between AT and pyAT (due to the transpose) a little tedious.

lfarv, 2017-08-31:
to @willrogers:

I'm back to the array ordering problem: I successfully installed the last pyat version from Github, and I tried the "physics" functions. `find_m44` returns a transfer matrix transposed compared to the usual way we see it in the books... I think we cannot leave it like that. So I go back to your proposals:

- Don't use `atpass` directly. This is the case in Matlab, where it's accessed only through `ringpass` and `linepass`. Similar functions in python could check and if necessary convert the input coordinates, to enforce Fortran-alignment at the input to `atpass`. This should be the only point where ordering matters.

- "physics" functions like `find_m44` could build their input to `ringpass` in the "normal" shape (6 X Nparticles), but specifying explicitly Fortran-ordering, so that they don't loose efficiency. Then, no transposition is necessary and the algorithm should stay the same as in Matlab.

- The user should use the standard orientation of matrices and not care about ordering, numpy takes care of it.

willrogers, 2017-08-31:
Yes, I agree with this proposal.  I'll try to find some time in the next few days to look at doing this.

lfarv, 2017-08-31:
I'll make a trial right now, (branch matrixordering), and we'll see what it implies !

willrogers, 2017-09-12:
Fixed!

closed 2017-09-12 by willrogers