atcollab/at #15: phase advance computation at given index

simoneliuzzo, 2017-09-28:
in 
`[l,t,c]=atlinopt(ring,0,indexes_in_ring)`

the output l.mu is not correctly computed.

To obtain the correct computation it is needed to issue the following comands:
```
[l,t,c]=atlinopt(ring,0,1:length(ring)+1)
l(indexes_in_ring).mu
```

lnadolski, 2018-03-10:
Is it corrected ? If not I do not understand the issue ?

simoneliuzzo, 2018-03-12:
the issue is still unresolved up to my knowledge.
`atlinopt `does not return the correct phase advance if only specific locations are asked. 
`[l,t,c]=atlinopt(r,0,1:length(ring)+1)` gives the correct `l.mu` at all elements
`[l,t,c]=atlinopt(r,0,   length(ring)+1)` does **not** give the correct ` l.mu` at the last element if the phase advance of ring is more than 2pi.

lfarv, 2018-03-12:
I do not see anything to do: if you want to right phase advance, be sure to set enough locations so that there is no locations such that the phase progresses by more than 2pi in-between. It would be very inefficient for most uses to have `atlinopt` compute at all points and later extract the requested points of interest. You can do it manually as proposed by Simone in his 1st comment if you need it.

simoneliuzzo, 2018-03-12:
The problem historically raised from `atmatch`. 

When matching the total phase advance for ESRF EBS cell, the call to match `l.mu` at a given location (cell end) specified using `atvariablebuilder ` made the matching fail (no convergence) due to the above problem in `atlinopt`. The work arround ever since is the one mentioned above, calling with all indexes (could be optimized) between the lattice start and the desired location. This makes the matching slow, since that way all optics are computed at all locations rather than at the few specific ones (I remind that `atmatch` is optimized such that if different optics knobs are asked at the same locations, `atlinopt ` is called only once per independent location). 

The issue is open to look for a different algorithm for `atlinopt` that would provide the correct phase advance without knowing the optics of the elements before. I do not know if such algorithm exists, but `MAD ` and `elegant ` do not have this kind of problem.

lfarv, 2018-03-12:
It looks to me like a fundamental problem: you cannot measure the integer part of the tune with a single BPM (end of the turn) To get it properly you need to look all around the ring and sample it finely enough not to miss one integer. It's the same in simulation. So no other algorithm ! If you want to be fast, you can match the fractional tune (or phase advance modulo 2pi).

lcarver, 2023-10-19:
can this be closed?

simoneliuzzo, 2023-10-25:
no objection.

closed 2025-07-23 by swhite2401