atcollab/at #18: Create README

willrogers, 2017-10-11:
We should create a README file for the AT repo. If we use Markdown or reStructuredText it will display nicely on Github.

@carmignani @simoneliuzzo @bnash @lfarv

bnash, 2017-10-16:
Where should this file go?

willrogers, 2017-10-16:
The root of the directory. Almost all repos on Github have one - see https://github.com/google/protobuf

Github will render it when you visit the repo so it works a bit like a homepage.

In fact, I put one in the pyAT directory, you can see it here: https://github.com/atcollab/at/tree/master/pyat. But we should have one for the repo as a whole.

willrogers, 2017-10-16:
I created a basic readme based on the webpage http://atcollab.sourceforge.net/.

You can see what it looks like here:  https://github.com/willrogers/at/tree/readme

closed 2017-12-04 by willrogers