atcollab/at #19: Correct names and email addresses in Git history

willrogers, 2017-10-11:
This is a little tricky because it requires re-writing the Git history, so anyone with a clone of at will need to reclone. Instructions are here: https://help.github.com/articles/changing-author-info/

This is what Git currently thinks the commits are:

```
(venv3) will@wrmpb ~/c/a/pyat> git shortlog -s -n -e
   325  Laurent Farvacque <laurent.farvacque@esrf.fr>
   255  Boaz Nash <boaz.nash@esrf.fr>
    60  Simone Liuzzo <simone.liuzzo@esrf.fr>
    44  Laurent Nadolski <laurent.nadolski@synchrotron-soleil.fr>
    35  Nicola Carmignani <nicola.carmignani@esrf.rf>
    31  Will Rogers <wahrogers@gmail.com>
     8  yretan <yretan@atcollab>
     4  peacechang <peacechang@atcollab>
     4  ncarmignani <n.carmignani@gmail.com>
     4  xiahuang <xiahuang@atcollab>
     2  thperron2001 <thperron2001@atcollab>
     1  Nicola Carmignani <nicola.carmignani@esrf.fr>
     1  bnash <boaz.nash@gmail.com>
     1  Giovanna Alea <gio.alea@gmail.com>
```

willrogers, 2017-10-11:
@carmignani @lnadolski @bnash @simoneliuzzo @gioAlea @yrtan @lfarv could you please confirm the name and email address you would like to be stored in Git history? 

We also have contributions from `peacechang`, `xiahuang` and `thperron2001`. Can we contact them?

This should also be configured for Git as described here: https://git-scm.com/book/en/v2/Getting-Started-First-Time-Git-Setup so that future commits match up too.

lfarv, 2017-10-11:
I don't like the idea of rewriting the history: the addresses are as they were when the commit was done, and it makes sense. I don't think you'll rewrite the whole history every time someone changes his mail address !
Now everybody has to move to git, so he will put in his config the most recent address, and all new commits will be up-to-date.
You also cannot ask everybody to reclone all the repositories. Though it is not what the instructions say (fetching is easier than cloning again): 
> "any person with forks or clones must fetch the rewritten history..."

So please let the history as it is, we know all those people...

bnash, 2017-10-11:
@lfarv, we had a short email exchange about this. I understand the point is to make a correction on an error in Nicola's email (.rf instead of .fr) and also to correct any other emails during this transition between sourceforge to github, so that everyone's contributions shows up here
https://github.com/atcollab/at/graphs/contributors
It would be a one time change as a part of the transition from Sourceforge to GitHub.  We could send an email to the mailing list to tell people what to do. 

I don't know if Peace, Xiaobiao or Thomas have github accounts.  The email addresses I have for them are Peace: peace@nsrrc.org.tw, Xiaobiao: xiahuang@slac.stanford.edu and Thomas: thomas.perron@esrf.fr

For myself, the esrf.fr email won't last much longer, so I prefer the gmail.com address.

willrogers, 2017-10-11:
In general it is not advisable to change Git history. You can fetch the changed repository but the old master branch and the new one will be different, so the easiest way to manage it is to re-clone. Anyone who has a branch in progress has to be careful.

The problem is that I had to provide the email addresses at the time of converting from Subversion, and I made a mistake in Nicola's and provided dummy addresses for everyone I didn't know of.

We would do this once only. In any case, I do not really have an opinion and I am happy with what the agreed consensus is.

lfarv, 2017-10-12:
I am still not convinced of the necessity to rewrite Git history. There are ~ 8 clones here at ESRF, some used for the machine operation, and I do not feel comfortable saying to some people (not git experts !) that they have to start again from zero. I have also here several local branches open… We moved away from SourceForge hoping to have something more stable, so please keep it such. Anything you do should be transparent for users.

Concerning the wrong email adresses, I don't see the problem: every user can easily update his profile on the Github site. These adresses will frequently change: Boaz's now, mine in ~2 months… Only the most recent address is relevant, addresses associated with old Subversion commits don't really matter.

So I strongly insist on keeping the repository stable: it's too late, the switch to Github has already occurred, at least at ESRF

lnadolski, 2017-10-12:
Dear All,

I agree with the opinion of Laurent F.

Best regards,

Laurent.

Dr Laurent S. NADOLSKI
Coordinateur Exécutif des Accélérateurs
Division Accélérateurs et Ingénierie, Synchrotron SOLEIL
t:      +33 (0)1 69 35 98 08  m: +33 (0)6 89 43 70 45
a:      L'Orme des Merisiers Saint-Aubin BP 48, 91192 Gif-sur-Yvette CEDEX FRANCE
w:      www.synchrotron-soleil.fr<http://www.synchrotron-soleil.fr>  e: nadolski@synchrotron-soleil.fr<mailto:nadolski@synchrotron-soleil.fr>

yrtan, 2017-10-13:
Newbie to git. My profile does have my correct email. As for history, as long as the git username is valid that should be enough for people to contact the person that made the change if necessary.

bnash, 2017-10-13:
At the moment, I'm not actively developing with this repository, so it doesn't bother me to make the change.  But if those who are using this repository are not comfortable with it, then I can agree not to.

willrogers, 2018-04-17:
I apologise to Nicola, but I think it has been agreed that this is no longer possible.

closed 2018-04-17 by willrogers