atcollab/at #30: Compatibility with Matlab 2017b

SA100, 2018-01-05:
I get --- AT was not installed correctly Installed in Matlab version 2017b.
Any compatibility issue? I used older 1.2 Version AT and also 2.0. 
ATRoot='E:\Accelerator\AT1.2'

ATRoot =

    'E:\Accelerator\AT1.2'

>> unzip('atzip.zip',ATRoot)
>> cd(ATRoot)
>> atpath
>> atdiag
>> spear3;
>> plotbeta
AT was not installed correctly
Thank you.
-sa

bnash, 2018-01-05:
Hi @SA100, can you give us the output of atmexall?  The main question is whether the mex functions get built correctly.

SA100, 2018-01-05:
Hello,
Please find atmexall output below:
>> atmexall

ATROOT =

    'E:\Accelerator\AT12'

Current directory: E:\Accelerator\AT12\simulator\element
mex -DPCWIN64 AperturePass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 BendLinearPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 BndMPoleSymplectic4E2Pass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 BndMPoleSymplectic4E2RadPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 BndMPoleSymplectic4Pass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 BndMPoleSymplectic4RadPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 CavityPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 CorrectorPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 DriftPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 EAperturePass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 IdentityPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 Matrix66Pass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 QuadLinearFPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 QuadLinearPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 SolenoidLinearPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 StrMPoleSymplectic4Pass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 StrMPoleSymplectic4RadPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 ThinCavityPass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
mex -DPCWIN64 ThinMPolePass.c 
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
Current directory:E:\Accelerator\AT12\simulator\track
mex -DPCWIN64 atpass.c
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
Current directory:E:\Accelerator\AT12\atphysics
mex findmpoleraddiffmatrix.c
Building with 'Microsoft Visual C++ 2015 (C)'.
MEX completed successfully.
ALL mex-files created successfully

bnash, 2018-01-05:
Ok, good, so the mexfunction look like they are being created correctly.

Any ideas for what to look for next, @lfarv or @carmignani or @simoneliuzzo ?

bnash, 2018-01-05:
I guess the thing to check is the map file.  The integrators need to be able to find the entry points into the mex files (mexfunction, trackfunction or passfunction...)

SA100, 2018-01-05:
I think part of map file is ok but if I go back to 2.0 version, file talk.m output file after mex files are compiled gives the following with errors. Version 1.2 is not compatible as far as I can tell.
----
>> talk
 spear2 example lattice definition file
  Created 11/21/99 
  Simplified SPEAR-II lattice
  no BPMs, no correctors

 spear2rad example lattice definition file
  with CAVITY and CLASSICAL radiation
  Created 11/21/99 
  Simplified SPEAR-II lattice
  no BPMs, no correctors

sp3v81loco not found.

Use the Help browser search field to search the documentation, or
type "help help" for help command options, such as help for methods.

 
** Loading SPEAR lattice in spear2.m **
** Done **
  Name            Size             Bytes  Class     Attributes

  FAMLIST         1x27             58032  cell      global    
  GLOBVAL         1x1                372  struct    global    
  QDW             1x1                  8  double              
  QFW             1x1                  8  double              
  THERING       271x1             373852  cell      global    
  kn1             1x20             12200  struct              
  kn2             1x10              6260  struct              
  qdlist          1x10                80  double              
  qflist          1x20               160  double              
  weights1        1x20              2400  cell                
  weights2        1x10              1200  cell                

ans = 

  struct with fields:

       FamName: 'AP'
    PassMethod: 'AperturePass'
        Length: 0
         Class: 'Aperture'
        Limits: [-0.0500 0.0500 -0.0500 0.0500]

ans = 

  struct with fields:

       FamName: 'DR01'
    PassMethod: 'DriftPass'
        Length: 1.3448
         Class: 'Drift'

ans = 

  struct with fields:

        FamName: 'Q3'
     PassMethod: 'QuadLinearPass'
         Length: 1
          Class: 'Quadrupole'
              K: 0
       PolynomB: [0 0]
       PolynomA: [0 0]
       MaxOrder: 0
    NumIntSteps: 10

ans = 

  struct with fields:

       FamName: 'DR02'
    PassMethod: 'DriftPass'
        Length: 0.8600
         Class: 'Drift'

ans = 

  struct with fields:

        FamName: 'Q2'
     PassMethod: 'QuadLinearPass'
         Length: 1.3427
          Class: 'Quadrupole'
              K: 0.0790
       PolynomB: [0 0.0790]
       PolynomA: [0 0]
       MaxOrder: 1
    NumIntSteps: 10

ans = 

  struct with fields:

       FamName: 'DR03'
    PassMethod: 'DriftPass'
        Length: 6.4132
         Class: 'Drift'

ans = 

  struct with fields:

        FamName: 'Q1'
     PassMethod: 'QuadLinearPass'
         Length: 0.5183
          Class: 'Quadrupole'
              K: -0.2596
       PolynomB: [0 -0.2596]
       PolynomA: [0 0]
       MaxOrder: 1
    NumIntSteps: 10

ans = 

  struct with fields:

       FamName: 'DR04'
    PassMethod: 'DriftPass'
        Length: 0.6119
         Class: 'Drift'

ans = 

  struct with fields:

          FamName: 'BB'
       PassMethod: 'BendLinearPass'
           Length: 2.3579
            Class: 'Bend'
     BendingAngle: 0.1848
    EntranceAngle: 0.0924
        ExitAngle: 0.0924
                K: 0
         PolynomB: [0 0]
         PolynomA: [0 0]
         MaxOrder: 0
      NumIntSteps: 10

ans = 

  struct with fields:

       FamName: 'DR04A'
    PassMethod: 'DriftPass'
        Length: 0.6171
         Class: 'Drift'

ans = 

  struct with fields:

     FamName: 'AP'
     NumKids: 1
    KidsList: 1
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR01'
     NumKids: 4
    KidsList: [2 136 137 271]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR02'
     NumKids: 4
    KidsList: [4 134 139 269]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR03'
     NumKids: 4
    KidsList: [6 132 141 267]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR04'
     NumKids: 16
    KidsList: [8 32 48 58 80 90 106 130 143 167 183 193 215 225 241 265]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR04A'
     NumKids: 4
    KidsList: [10 128 145 263]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR05'
     NumKids: 4
    KidsList: [12 126 147 261]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR06A'
     NumKids: 4
    KidsList: [14 124 149 259]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR06B'
     NumKids: 4
    KidsList: [16 122 151 257]
    ElemData: [1×1 struct]

ans = 

  struct with fields:

     FamName: 'DR07A'
     NumKids: 4
    KidsList: [18 120 153 255]
    ElemData: [1×1 struct]

MAD output file: spear3_mad_structure.out
 
MAD file type:           STRUCTUR
Symmetry flag:           0
Number of superperiods:  1
Number of elements :     790
 
Warning: MAD lattice contains RF cavities 
> In readmad>mad2at (line 136)
  In readmad (line 101)
  In talk (line 17) 
 
AT cell array was successfully created from MAD output file spear3_mad_structure.out
Some information may be not available in MAD otput files
Some elements may have to be further modified to be consistent with AT element models
 
For RF cavities READMAD creates elements that use DriftPass or IdentityPass (if Length ==0)
Use CAVITYON(ENERGY) [eV] in order to turn them into cavities
 
** Loading SPEAR lattice in spear2.m **
** Done **

ans = 

  struct with fields:

        FamName: 'Q3'
     PassMethod: 'QuadLinearPass'
         Length: 1
          Class: 'Quadrupole'
              K: 0
       PolynomB: [0 0]
       PolynomA: [0 0]
       MaxOrder: 0
    NumIntSteps: 10

Reference to non-existent field 'R1'.

Error in talk (line 35)
THERING{3}.R1

bnash, 2018-01-05:
Ok, so you are saying that AT is actually working in general, but this particular example isn't working?  Can you get tracking to work with some of the other lattice files in
https://github.com/atcollab/at/tree/master/machine_data

bnash, 2018-01-05:
Try some of the demos in
https://github.com/atcollab/at/tree/master/atmat/atdemos

SA100, 2018-01-05:
Thank you so much for your time.
here is the output of latticedemo.m -- missing mex ???
% Lattice THERING is a variable in MATLAB workspace.
% We can use it in accelerator physics functions and scripts
%
% For example: function FindM44 finds 4-by-4 transverse transfer matrix
M = findm44(THERING,0)
Error using atpass (line 28)
missing MEX file.

Error in linepass (line 97)
    [Rout,lossinfo] = atpass(line,Rin,newlattice,1,refpts,prefunc,postfunc,nhist);

Error in findorbit4 (line 75)
    RMATf = linepass(RING,RMATi,args{:});

Error in findm44 (line 86)
    R0 = [findorbit4(LATTICE,DP);DP;0];

Error in latticedemo (line 125)
M = findm44(THERING,0)
 
>>

bnash, 2018-01-05:
Yes, so it does look like the integrator can't find the right entry point in the mex file.
You should try as simple a function as possible when debugging.  Load a lattice (e.g. dba.m or esrf.m) and then run atlinopt(ring) or findm44(ring,0) or findm66(ring), something like that.

In case the basic tracking (ringpass) isn't working, then it probably is an issue with the map file that tells Matlab which entry points to look for in the mex functions.

bnash, 2018-01-05:
The simplest thing to try is to create an element, say using atquadrupole, then try StrMPoleSymplectic4Pass using the quadrupole as an argument.  This will use the mex function entry point to the integrator.
(Sorry, at the moment I don't have Matlab set up on this computer, so I'm less useful...)

bnash, 2018-01-05:
"missing MEX file" also suggests something about your path, or where the mex functions go.  Can you check that the mex files were actually put in the atintegrators directory?  Maybe you need to move them after the atmexall command.

carmignani, 2018-01-08:
Hello,
that "missing MEX file." is an error of atpass, printed by atpass.m, so it looks like atpass.mex is not in your path. Can you check if the compiled file of atpass.c has been written in the directory at/atmat/attrack?
You should have there atpass.c, atpass.m and atpass.mexa64
Try also to type which atpass. You should have something like this:
/users/carmigna/matlab/AT2/at/atmat/attrack/atpass.mexa64

lfarv, 2018-01-08:
Hello,
I also think that the mex-files are missing the right entry points. There was an issue when Matlab >= 2016b changed the mex-file compile method. It has been corrected for linux and macOS by updating the arguments of the `mex` command, but I don't know for Windows. I could get back a virtual Windows 7 machine running, but it's on Matlab 2015b, and AT runs correctly. Has anybody experience of 2016b on Windows ? Anyway I'll try now  to install 2017a and make tests.

bnash, 2018-01-08:
@yrtan Do you have experience with AT with Matlab 2017b on Windows, as asked by @lfarv?

@SA100 , any progress on figuring out the issue here?

yrtan, 2018-01-09:
Compile issue was only for Linux. Windows doesn't appear to have the same symbol issue where the error was "No passFunction or trackFunction available". 

The "missing MEX" does appear to be a path issue. I suggest downloading [2.0](https://github.com/atcollab/at/releases/tag/2.0), to a clean install by removing all other versions and trying again --> atpath; atmexall; atdiag;

If the problem persists show us the result of >>path just to make sure the path is correct and points to atpass.mexw64 or atpass.mexw32 (depending on your OS).

=== WINDOWS (what I have currently) ===
AT1.??? (don't really know what version, soooo old) + Matlab 2016a is **ok**.
AT1.4.1 + Matlab2016a is also **ok**.

SA100, 2018-01-09:
For your info:
I remove the downloaded zip version from my computer and used "git clone". Path and installation is identical to the zip version but now everything works fine. Thank you for your help. -SA

closed 2018-01-09 by SA100

bnash, 2018-01-09:
Good to hear it @SA100 .
Thanks for contributing here.  Feel free to raise other issues.  More people discussing creates a good environment to keep the code and the collaboration alive.