atcollab/at #40: nafflib problems

carmignani, 2018-03-13:
Hello,
I tried to use the calcnaff function that we have in the nafflib directory and I found some problems.

If I run the function to analyze the same signal two times I get the same frequencies, but the phases are every time different.

A second issue about the function is that it requires a complex signal, so it cannot be used with real bpm measurements. We could give a vector of zeros as imaginary part of the signal, but then I don't know if we can trust the results.

yrtan, 2018-05-14:
Tried the default example from calcnaff and got the same phases each time. Also tried it on some BPM data. I couldn't repeat the problem for AT2.0. nafflib.c revision date 25/7/2017.

You can use it for just real signals and the phases be also be correct (taking only the positive frequencies). However for the amplitude you have to sum the positive and negative frequency amplitudes to get it right. Or you can simply multiply by a factor of 2 if the frequency is not zero or fs/2.

lnadolski, 2020-07-27:
Dear @carmignani, 
I cannot reproduced your bug.

If you run the nafflib algorithm with a real signal, its works well for me.
This is a simple example :

> t = linspace(1,2*pi*20,1000);

> [frequency amplitude phase] = nafflib(sin(t),sin(t)*0, 1, 5, 1)
*** NAFF results ***
NFS = 2
AMPL= 4.207309e-01+i* 2.701499e-01 abs(AMPL)= 4.999955e-01 arg(AMPL)= 5.707992e-01 FREQ=-1.247885e-01
AMPL= 4.207355e-01+i*-2.701512e-01 abs(AMPL)= 5.000000e-01 arg(AMPL)=-5.707963e-01 FREQ= 1.247885e-01

The phase is stable and the amplitude of each harmonic is 0.5 as expected.

lnadolski, 2020-07-27:
Nevertheless, I suspect **something wrong the in the c-library**, since sometimes, the function returns NaNs for valid input signal.
If I rerun again the same command with the same input argument the results is then fine.
I suspect something not properly defined with the dynamic memory allocation in the code.

I need to run the code in debugger to understand. This is quite annoying and time consuming to debug.
Any advice is very welcome.

carmignani, 2020-07-30:
I get exactly the same output running the example you wrote, but if I run it many times sometimes I get nans. 
I also suspect there are some problems in the dynamic memory allocation in the C code.

PR #587 (2023-03-29): Fixes bug producing infrequent NaN from calcnaff
Dear all,

This pull request concerns MATLAB.
As noted in #40 calcnaff sometimes produces NaN values even though the input is valid and a second run does result in valid frequencies. 
The problem is that the interface between MATLAB and the NAFF algorithm did no write to the last element of the
complex input to NAFF.
This pull request fixes that.

In my tests this also fixed the fluctuating results from run to run (e.g. phase) seen in #40 as well.

Fixes #40
commit 5c287fd9 (2023-03-29): Fixes bug producing infrequent NaN from calcnaff (#587)

* Fixes bug producing infrequent NaN from calcnaff

Fixes #40

* Update calcnaff.m

The second test for NaN was wrong Y was replaced by Yp

---------
closed 2023-03-29 by lfarv