atcollab/at #44: recycling pass methods

ZeusMarti, 2018-03-21:
Hi all,

I'm starting to rewrite some of our AT1.3 pass-methods to AT2.0:

(A). A linear version of IdTable,  It is defined by 4 parameters, dx'/dx, dx'/dy,dy'/dx and dy'/dy, it is useful to fit Id effects via its RM changes. We also used it to fit transverse impedance effects in the RM.
(B). An Sliced version of BndMPoleSymplectic4Pass. It uses PolynomA and PolynomB defined as matrices (number of coefficients X number of slices). The rad version is also done, but the  ringpara and omhienvelope modifications are pending.
(C). A full Hamiltonian version of BndMPoleSymplectic4Pass, StrMPoleSymplectic4Pass and DriftPass. My idea was to be able emulate MAD's "Exact" flag.

Please, tell me if you think it would be useful to have any of them in this repository!

Cheers,

simoneliuzzo, 2018-03-21:
Hi Zeus, 
this sounds very interesting! We where starting to implement (B) and have a version of (C) for Drifts only (DriftExactPass, not in the repository, but tested). 
Will you propose additional passmethods or modify the existing ones with some option flags?
ciao
simone

ZeusMarti, 2018-03-21:
Hi Simone,

Ok, then I'll try to translate first (A). Regarding (B) and (C), I wrote them several years ago, I got the formulas from Forest's book, I did tested them, but not super-exhaustively. I can pass you the code in the AT1.3 format if you want. I'm not sure how long it could take for me to translate them so it may be faster this way. Currently they are separated passmethods, I guess that modifying existing versions should be done more carefully, is there any preferred option?

Best,

willrogers, 2018-03-21:
At DLS we have `ExactHamiltonianPass.cc`. This was implemented by a previous employee who was very reliable, but I think you would have to do testing. I believe it would handle a generic bend+multipole magnet.

It's also written for AT 1.3 but I could update that.

carmignani, 2018-03-21:
Hello Zeus,
the AT1.3 format should be perfectly compatible with AT 2.0. The only problem is that you could not use them in python AT. Instead of sending the passmethods by email, you could create a branch in github and put them there.

ZeusMarti, 2018-03-21:
Ok Nicola, I think I could do that, but I'm a git beginner. I have an appointment next week with the guy who gave us a git crash-course some months ago. Now that I have a clear motivation to use it, I think I'll could start soon.

Will, if we add the codes in the same branch we could test them together and decide what to do. Does it make sense?

bnash, 2018-03-21:
Hi all,
For testing integrators, I put a very primitive testing approach here:
https://github.com/atcollab/at/blob/master/atmat/atdemos/tracking/
Basically, one could use this to check and see that the tracking results don't change when they aren't expected to (e.g. if a new flag is added to an integrator).  
The idea is to:
*)Use old integrators to generate tracking data on a number of lattices with
https://github.com/atcollab/at/blob/master/atmat/atdemos/tracking/generateTrackData.m
*)make the changes to the integrators, 
*)then run 
https://github.com/atcollab/at/blob/master/atmat/atdemos/tracking/testTracking.m
to see if anything changed.

bnash, 2018-03-21:
@ZeusMarti, One easy way to get going with git and github is using the [GitHubDesktop](https://desktop.github.com/) program.  I found that useful recently.  Its pretty easy to create branches with it, and then push that to the atcollab repository.

On your passmethod (A), how is this different from a matrix?  Could you just use the matrix66 pass method for this?
https://github.com/atcollab/at/blob/master/atintegrators/Matrix66Pass.c
Probably I'm missing something.

Sounds like we have two different versions of the exact dipole pass method, one from @ZeusMarti Zeus, and one from @willrogers (was it written by James Rowland?).  I don't know the best way to move ahead, but I think that getting the code into branches here on github is a good start so we can look at the code, try it out, and offer suggestions.

In general, if we can generalize or improve the pass methods we already have, I think that's better than adding a lot of different versions that work slightly differently.  But if a pass method provides something that doesn't already exist, then I think its worth adding.

ZeusMarti, 2018-03-22:
Hi Boaz, sorry, I just did not notice Matrix66Pass looks similar to what I was proposing. After quick check, the main difference I see is that method (A) is meant for long devices, it divides the pass into symplectic drift-kick-drift slices. I guess Matrix66Pass could be used for that purpose too, via concatenation with drifts or directly writing the matrix elements to make it both symplectic and including the drift, but I think that is not immediate, specially for the coupling part. But I agree with you, redundancy should be avoided, I won't include it if no one asks for it.

Regarding GitHubDesktop, I was not able to install it, the installation log says something weird to me:

2018-03-22 07:17:26> CheckForUpdateImpl: Couldn't write out staging user ID, this user probably shouldn't get beta anything: System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\Users\zeus\AppData\Local\GitHubDesktop\packages\.betaId'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)

Next week I'll dedicate some time to it!
Cheers,

ZeusMarti, 2018-03-27:
Hi all,

Can someone create a branch so that I can add the passmethods? I can not see Will's 'ExactHamiltonianPass.cc' branch containing his coleague's version. Otherwise, I uploaded a [repo](https://github.com/ZeusMarti/Full-H-passmethods.git) with only the new/changed files:

- Sliced &/| Full Hamiltonian versions of Str and Bnd methods. 
- An atphyslib.c version with some more auxiliar functions.
- A findmpoleraddiffmatrix.c version that also accepts sliced elements (not HFull yet).

I'm also working to modify atsummary and ringpara, 

I hope it is helpful,

simoneliuzzo, 2018-03-28:
Hi Zeus, 

I think that the correct procedure is that you "`fork`" the project and make a `pull request` for the `fork `modified. Am I correct @willrogers ?

ciao
Simone

carmignani, 2018-03-28:
I don't understand why you should fork and not make a branch. Maybe @willrogers can explain.
In my opinion a branch is better, because then everybody could commit there.

ZeusMarti, 2018-03-28:
I don't have permits to create a brach in atcollab/at. I have already forked /at in my github and added a new brach there, but then, where my pull request should target? I was assuming that it would be better it to target a new branch in atcollat/at rather than to the master...

Sorry if I'm missing something, I'm just a beginner. Any way, I'll be on vacations for few days, I'll try to catch you up after easter!

willrogers, 2018-03-29:
The advantage of a 'fork' is that the user doesn't need permission to the main repo. In our case it doesn't really matter. Should someone add @ZeusMarti to atcollab?

bnash, 2018-03-29:
I sent an invite to add @ZeusMarti to atcollab.

ZeusMarti, 2018-04-09:
I just added a new branch. Two functions are modified: findmpoleraddifmatrix and atphyslib, the rest are just new pass-methods without the python compilation (sliced elements and full H). Some more explanation in the commit. work to do:
- Compare full Hamiltonian methods
- adapt ringpara and atsummary 

Collaborate is wellcome!

closed 2025-07-23 by swhite2401