atcollab/at #50: atlinopt off energy

carmignani, 2018-04-18:
If we compute the linear optics with atlinopt with the RF frequency changed we would expect to get the off-energy optics, but this is not the case now. atlinopt uses always findorbit4 (through findm44) to get the closed orbit.
The only way to get the off-energy optics is to pass the parameter dpp.
What do you think about using findorbit6 to get the closed orbit and then computing the linear optics around this orbit? In this case, we could compute the off-energy optics just by changing the RF frequency.
We could have a flag, something like "UseOrbit6" or "UseRFFrequency". If the flag is present, the second input of atlinopt would be ignored, the closed orbit would be computed with findorbit6 and the orbit would be passed to findm44 to get the optics.

simoneliuzzo, 2018-04-22:
this sounds good to me. using the flag allows to keep the previous usage unchanged. (i'd like a simpler name, like 'rf_priority', '6Dorbit', 'frequency' )

lfarv, 2018-04-22:
I see a problem with that idea: `atlinopt` does a 4x4 processing. It uses `findm44` which says:
 
>   IMPORTANT!!! findm44 assumes constant momentum deviation.
>     PassMethod used for any element in the LATTICE SHOULD NOT
>     1.change the longitudinal momentum dP
>       (cavities , magnets with radiation, ...)
>     2.have any time dependence (localized impedance, fast kickers, ...)

(By the way I think that `atlinopt` help should also give the same warning)
Some of the `atlinopt` output make no sense in 6x6 processing (dispersion,…).
What you propose is a mixture, where the orbit would be computed in 6x6, then cavities and radiation would be removed (but then how to transport the initial closed orbit ?), and then a 4x4 processing would be done…

Don't you think that another function for full 6x6 treatment would make it more clear?