atcollab/at #56: pyat: data ordering in 2-D element parameters

lfarv, 2018-07-16:
While converting the last integrators to the Python-compatible `trackFunction`, I noted some problems with incompatible data ordering in element parameters.

Data ordering for the coordinate arrays (Rin/Rout) has been decided some time ago ( #6): 6xN matrices with Fortran-ordering. This is transparent for the user since `lattice_pass` converts if necessary to the correct ordering.

But for element parameters, the access functions (`atGetDoubleArray`...) check that the parameters are in C order (I do not remember why). Fortunately, 2-D parameters are quite rare: I found:

- Interpolation tables for the `IdTablePass` integrator (newly converted)
- 1st and 2nd order transfer matrices for the `Matrix66Pass` and `MatrixTijkPass` integrators
- `R1` and `R2` axes rotation matrices for all elements: **here I think the present implementation is buggy**

Again we have 2 solutions for clarifying this:

1) Keep element parameters in C-order: this means that the C code must be different for Matlab and Python when using 2-D parameters (using `#ifdef`s). This is what I did for the interpolation table in `IdTablePass`.

2) Decide that internally, all element parameters in AT are Fortran-ordered. C code is then identical for all versions and data conversion to/from Matlab is easier. Again this can be made almost transparent for the user if elements are built by element creation functions (`element.py`). This choice concerns only arrays accessible from C-code: particle coordinates and element parameters. For everything else, data ordering is handled transparently by numpy. This is my preferred solution.

As soon as a decision is made, I can take care of atelem.c and the few concerned integrators.

willrogers, 2018-07-16:
I tend to agree: use option 2. There is the potential for some 'advanced' user to get this wrong in python (is there anywhere that we can check and catch this at runtime?) but it is most consistent and would be simpler in the C code.

carmignani, 2018-07-16:
In the branch fieldmaptracking there are two new passmethods using a lot of 2-D arrays.

lfarv, 2018-07-16:
@willrogers : Access to the parameters in the the integrators is done through a utility function like `atGetDoubleArray` or so, which checks the type (double,...) and ordering of the data. So it is reasonably safe.

@carmignani : the ordering problem only concerns Python. Nothing needs to be changed in Matlab, and in option 2, the code will also work in Python without any change. To use `mxGetDoubles` from now on, you need the new `atelem.c`  which checks the Matlab version in use.

closed 2025-07-23 by swhite2401