atcollab/at #63: AT_2_OPA file-saving error

JonasKallestrup, 2018-08-17:
I experience a problem when using AT_2_OPA. OPA throws an error when loading the file. Example:
    spear3;
    AT_2_OPA(THERING,'test_AT2OPA');

Then loading test_AT2OPA_lattice.opa in OPA leads to the error:
"LATTICE ERROR > undefined entry RF in segment test_AT2OPA"

Though, there is a fix! If I manually open test_AT2OPA_lattice.opa in e.g. notepad++, copies all the text, and insert it into a new file with name e.g.  'newTest_AT2OPA_lattice.opa' there seem to be no problem loading the file in OPA. 
This suggests that saving the file in Matlab makes it unavailable in OPA, but a file created by notepad++ is OK even though that the content of the two files are identical. 

Has anyone seen something similar? 
Tested with MATLAB R2018a and R2016b.

lfarv, 2018-08-19:
I have no way to make tests, but I suspect that the problem comes from line termination conventions. May be OPA needs Windows-terminated lines (Cr-Lf) while Matlab uses Unix-terminated lines (Lf). Going through Notepad fixes that.
One may try in `at_2_opa.m` to open open the file in text mode:
`of=fopen(outfile,'wt')` instead of `of=fopen(outfile,'w')`.
This should work if the OPA file is generated by Matlab on Windows. Otherwise I do not know how to force Matlab to generate Windows-terminated lines.

JonasKallestrup, 2018-08-20:
Thanks @lfarv - using fopen(outfile,'wt') fixed the problem!
Though, there seem to be some discrepancies between the calculated optics in AT vs. OPA (the spear3 
 optics are wrong, while e.g. ASTRID optics are correct). 
I'll see if I can find the error, fix it if possible, and then do a pull-request.

closed 2018-08-20 by JonasKallestrup