atcollab/at #65: Problem with findm44

willrogers, 2018-08-29:
I did the following:

```
atpath
load atdemos/atmatchExamples/ExampleATMATCH/dba.mat
findm44(RING, 1e-5)
```

I get the following error:

```
Error using  + 
Matrix dimensions must agree.

Error in findm44 (line 120)
RIN = R0(:,ones(1,9)) + [D4 -D4 zeros(6,1)];
```

Is this a bug?

carmignani, 2018-08-29:
Hello,
I just tried and I get this:

ans =

   -0.6638    2.2341         0         0
   -0.2504   -0.6638         0         0
         0         0   -0.9992    0.2622
         0         0   -0.0059   -0.9992

It looks like you have a different version of findm44, because "RIN = R0(:,ones(1,9)) + [D4 -D4 zeros(6,1)];" is in line 118, not 120.

willrogers, 2018-08-29:
Sorry, looks like it must be a local error then.

closed 2018-08-29 by willrogers