atcollab/at #66: sign of b1 multipole in quad (sextu..) wrong?

lnadolski, 2018-08-31:
Dear All

I am facing something strange.
I set b1 to some value in a quad (or sextupole).
The orbit generated is of opposite sign of what is expected.
In the vertical plane this is fine.

I add the testing code using ESRF lattice.

Please let me know if you confirm the bug or if I am doing something wrong.

%% ESRF Lattice
Ring=esrf;

[LinData,nu, ksi] = atlinopt(Ring,0,1:length(Ring)+1)
beta = cat(1,LinData.beta);
phase= cat(1,LinData.mu);
eta  = cat(2,LinData.Dispersion)';
spos = cat(1,LinData.SPos);

Id = 6;
theta = 1e-4;
orb_ana = (sqrt(beta(Id,1)*beta(:,1))/2/sin(pi*nu(1)).*cos(abs(phase(Id,1)-phase(:,1))-pi*nu(1))-eta(Id,1)*eta(:,1)/mcf(Ring)/LinData(end).SPos)*theta;

Ring{Id}.PolynomB(1)=theta/Ring{Id}.Length;

cod = getcod(Ring);
cod = cod';

figure(400)
clf
subplot(2,1,1)
plot(spos, orb_ana); hold all;
plot(spos, cod(:,1),'k*');
legend('Analytical', 'AT')
ylabel('horizontal orbit')

orb_ana2 = (sqrt(beta(Id,2)*beta(:,2))/2/sin(pi*nu(2)).*cos(abs(phase(Id,2)-phase(:,2))-pi*nu(2))-eta(Id,3)*eta(:,3)/mcf(Ring)/LinData(end).SPos)*theta;

Ring=esrf;
Ring{Id}.PolynomA(1)=theta/Ring{Id}.Length;

cod = getcod(Ring);
cod2 = cod';

subplot(2,1,2)
plot(spos, orb_ana2); hold all;
plot(spos, cod2(:,3),'k*');
legend('Analytical', 'AT')
ylabel('vertical orbit')

lfarv, 2018-09-02:
This is not a bug but a consequence of the field expansion in AT. You can think of a “standard” bending magnet : it bends to the right (negative angle), while if you rotate it by 90 deg. It bends to the top (positive vertical angle). So definitely, as noted by @carmignani:
Positive B1 means negative H angle
Positive A1 means positive V angle

oscarxblanco, 2024-08-09:
Dear all,
I have been checking the sign conventions, and the comments above by @lfarv and @carmignani seem correct for a particle with positive charge.

I couldn't find any place in the page (https://atcollab.github.io/at/common/indx.html) where this is stated. Could it be included ?
Besides that, does the sign of `charge` in the pyat class `particle` have any effect ?
https://atcollab.github.io/at/p/api/at.lattice.particle_object.html#module-at.lattice.particle_object