atcollab/at #67: Phase-space portrait with atfastring

JonasKallestrup, 2018-09-06:
Hello,
I am doing some tracking-related studies, which include painting the x-x' phase-space portrait, and noticed that a ring produced by atfastring leads to a significantly different portrait than the regular ring.
My test code:
> ring = esrf;
> [ring_fast,ring_fastRad] = atfastring(ring);
> x = 0:1e-4:5e-3;
> Rin = zeros(6,length(x));
> Rin(1,:) = x;
> Rout_fullRing = ringpass(ring,Rin,1000);
> Rout_fastRing = ringpass(ring_fast,Rin,1000);
> figure;
> subplot(1,2,1)
> plot(Rout_fullRing(1,:),Rout_fullRing(2,:),'b.');
> axis([-6e-3 6e-3 -1.5e-4 1.5e-4]);
> title('Normal ring portrait')
> subplot(1,2,2)
> plot(Rout_fastRing(1,:),Rout_fastRing(2,:),'r.');
> axis([-6e-3 6e-3 -1.5e-4 1.5e-4]);
> title('Fast ring portrait')

The resulting plot:
[image]

Is this a fundamental limitation of the nonlinear optics in atfastring, or is it fixable somehow?

carmignani, 2018-09-06:
Hello,
this is normal. With atfastring the nonlinear dynamics is not at all well described. Only the chromaticity and the tune shift with amplitude are included, but you can't see any phase space distortion. 
If you are interested on the tracking of particles at large amplitude, you should not use atfastring.

JonasKallestrup, 2018-09-06:
Okay, thanks!

closed 2018-09-06 by JonasKallestrup