atcollab/at #81: pyat: Matlab test failures

T-Nicholls, 2018-11-06:
@willrogers and I are experiencing failures of the new Matlab tests from #80. Initially on R2014b using Python 3.4, but they also fail for me on R2016a.
@lfarv what versions of Matlab and Python are you using?

T-Nicholls, 2018-11-06:
Of the 12 errors that we get 10 can be resolved by reverting commit f8e44f4c8387c3ad2687e29d55670c0c0241b5fb.
In atmat/atphysics/ParameterSummaryFunctions/atlinopt.m:
```
-  JUMPS = diff([0;P]) < -1.e-3;
+  JUMPS = [0; diff(P)] < -1.e-3;
```

lfarv, 2018-11-06:
I am using Matlab 2016a or Matlab 2018a on MacOS 10.14. I created a new working directory, I tried running on several lattices, with various refpoints, I could not reproduce you problem… Finally I tried Matlab 2016a on debian8, and now I got it ! This looks very strange, I'm looking at that right now !

lfarv, 2018-11-06:
It turns out that matlab behaves differently with empty matrices. On macOS it can add empty matrices with different sizes ( 0X1, 0x0, 1x0 for instance) while it fails on linux. I pushed a fix on #82. Can you test it?

T-Nicholls, 2018-11-07:
Interesting, I would not have expected such a difference between the same Matlab version on different platforms. #82 fixes all 12 of the failed tests, and everything behaves as expected now.

willrogers, 2018-11-07:
Just to note that pyat is now helping to catch potential Matlab AT bugs. I think this is a good thing!