atcollab/at #85: Physical constants sometimes change!

carmignani, 2018-11-09:
I noticed that the official value of some important physical constants stored in the file PhysConstant.m has been changed in the last years. 
For example, the elementary charge is now 1.6021766208(98)e−19 C according to wikipedia and apparently it will become _exactly_ 1.602 176 634e−19 C soon, but in our PhysConstant.m file the elementary charge value is only 1.602 176 487e-19 C!
I thought that we could just download the updated PhysConstant.m file, but then I noticed that the file is produced in Soleil and it does not come from NIST.
Should we modify the file?

lnadolski, 2018-11-09:
Hi Nicolas

The file is coming from  https://fr.mathworks.com/matlabcentral/fileexchange/24236-fundamental-physical-constants.
We just modified it to make it as a class.
We did no spend time to update it from the latest values from NIST.
Feel free to modify even if this will not change the results for our simulations.

Laurent.

lnadolski, 2020-07-27:
@carmignani 
can I close this issue?

carmignani, 2020-07-28:
Hello @lnadolski,
I never updated the physical constants file, but it is really not a problem for our simulations, because the error is in the 10^-7 range. So we can close the issue.

closed 2020-07-28 by carmignani