atcollab/at #86: MaxOrder is zero for magnets with zero strength

JonasKallestrup, 2018-11-12:
Creating an element that is turned off (PolynomB = 0 for quad, sext, etc.), the MaxOrder is automatically set to 0, since max([1 find(abs(elem.PolynomB)>0,1,'last') find(abs(elem.PolynomA)>0,1,'last')])-1 = 0.

If one wants to turn on the element, it is then not enough to make the PolynomB array non-zero; one must also change MaxOrder to the desired value manually, which seems like an unnecessary extra step, which can easily be forgotten. 

Instead, one could check the class of the element in atbaselem, and let MaxOrder = 1 for quads and 2 for sextupoles per default. I guess this may increase computation times somewhere (an element with MaxOrder = 1 is included in calculations, even though PolynomB(2) = 0), but if the number of turned-off elements is low, then it does not matter too much(?).

ZeusMarti, 2018-11-12:
Right, it happened to me once...

lfarv, 2018-12-07:
Solved by #93

closed 2018-12-07 by lfarv