atcollab/at #101: PyAT Matlab comparison test failures

T-Nicholls, 2019-03-12:
The new parameters comparison tests have been failing for me. All three failed at the mcf comparison with slightly different values between Matlab and Python:

| dp | py_mcf | ml_mcf |
| -- | ------ | ------ |
| 0.0 | 8.506395615730056e-05 | 8.506669244225616e-05 |
| 0.01 | 8.820482604883597e-05 | 8.820785279280231e-05 |
| -0.01 | 8.231552705045038e-05 | 8.23179738933576e-05 |

Is this just a tolerance issue, to what accuracy would they be expected to agree? For reference, I had failures, all with roughly the same values, on:

| Python | Matlab |
| ------ | ------ |
| 2.7.3 | R2014b |
| 3.3.7 | R2016a |
| 3.4.9 | R2016a |

lfarv, 2019-03-14:
Hi @T-Nicholls. Are you sure that:
- you use the last version of AT
- you call the python `get_mcf` with` ddp=1.E-6`, which is what Matlab uses
- you run on the same machine

I get an agreement with 15 digits for dp=0, and with 10 digits for dp!= 0
```
 0.00  8.506669244225616e-05
 0.01  8.820785279e-05
-0.01  8.2317973893e-05
```
Which agrees with your Matlab results !!!

(running on MacOSX 10.14, python 2.7 and 3.6, Matlab R2018a)

T-Nicholls, 2019-03-14:
@lfarv my apologies, a change I made caused the difference.

Ok, I have found the source of the problem:
- I was using a branch where the `py_hmba` fixture had a scope session, i.e.
    ```
    @pytest.fixture(scope='session')
    def py_hmba():
        return Lattice(load_mat(utils.hmba_ring, key='RING'), keep_all=True)
    ```
- We turn on radiation in the lattice in `test_ohmi_envelope`:
    ```
    # Python call
    py_lattice.radiation_on()
    py_emit0, py_beamdata, py_emit = physics.ohmi_envelope(py_lattice, refpts)
    ```
- The same lattice is used and the radiation is not turned off. Adding a line to turn it off fixes the issue:
    ```
    # test momentum compaction factor
    py_lattice.radiation_off()
    py_mcf = py_lattice.get_mcf(dp, ddp=1.E-6)  # Matlab uses ddp=1.E-6
    ```

Should we be ensuring the radiation state of the lattice, as we do for `linopt` and `ohmi_envelope`, or is `get_mcf` designed to be run in both states and the difference to be expected?

lfarv, 2019-03-15:
`get_mcf` is based on `find_orbit4`. Strictly speaking, this is valid if there is no longitudinal motion (radiation off), though many people use it in any condition (as you could see, the results are not very different). The `get_mcf`  function works for any sequence of elements and cannot be protected, but the `get_mcf` method of the `Lattice` object could be, as it is done for `linopt`.

T-Nicholls, 2019-03-15:
@lfarv thank you for the clarification, I will protect the `get_mcf` method on `Lattice` then.

closed 2019-03-15 by T-Nicholls