atcollab/at #111: How can I

windlessness, 2019-04-23:


windlessness, 2019-04-23:
Hello!

 I want to create some deviation in the ring' magnetic field as I am going to learn about the beam's closed orbit correction. 

However, I founded that the deviation in PolynomA[0] of dipole , PolynomB[0] and PolynomA[0] of quadrupole have not been considered during the physical calculation. 

The orbit haven't change though I add some deviation in PolynomA/B.

I wonder if my pass method has a great influence on it.

My dipole's pass method is BndMPoleSymplectic4E2Pass, and quadrupole's is QuadLinearPass

I am grateful and looking forward to the answer as a beginner of AT!

JonasKallestrup, 2019-04-23:
Hi,
Do you use the Matlab version of AT or the new (under development?) python version (pyAT)?

If Matlab:
The (in my opinion) best way to make a closed orbit distortion is by taking a corrector in the lattice (using _atcorrector_ to create the element), and then applying some kick. This is done by inserting some kick-angle in the .KickAngle field of the corrector element. 
A more advanced method would be to shift one of the quadrupole elements (using _atshiftelem_).

carmignani, 2019-04-23:
Hello,
PolynomA and PolynomB fields are ignored by the pass method QuadLinearPass. You should use StrMPoleSymplectic4Pass.

Jonas, you can add the field KickAngle to dipoles and multipoles too, you don't need to add a thin corrector element.

windlessness, 2019-04-23:
Thanks a lot, Mr. Jonas and Mr. armignani!  

My environment is pyAT. A senior in my lab told me that directly changing the PolynomA and PolynomB fields seem to work in Matlab AT， instead of changing the K value of dipole or quadrupole.

closed 2019-04-23 by windlessness