atcollab/at #117: Different modeemittances as calculated with atx.m for the same ring with different starting points!

z-duan, 2019-05-30:
Hi all,

      I calculated the mode-emittances of an imperfect ring lattice with atx.m, I tried to rotate the lattice to start with different elements (say, start the lattice at straight section centre, or at one arbitrary place in the arc), but the calculated mode-emittances(in particular the vertical emittance) diff for different starting points, which seems nonphysical. In the following lines I used the esrf lattice as an illustration:

RING=esrf();                                                                                                                                                                                                                  
ringpara(RING)
indS=find( atgetcells( RING, 'Class', 'Sextupole' )); 
s00=findspos(RING, 1:length(RING)+1);

[bd, par]=atx(RING,0,1:length(RING));
emits=par.modemittance;

RING1 = RING;
%%%%% misalign sextupoles to represent machine imperfections
dxs = 100e-6 * randn([length(indS),1]);
dys = 100e-6 * randn([length(indS),1]);

RING1=atsetshift( RING1, indS, dxs, dys);

[bd1, par1]=atx(RING1,0,1:length(RING1));
emits1=par1.modemittance;
fprintf('%.4e ',emits1);fprintf('\n');

nlen=length(RING);
results=zeros([nlen, 7]);
for ind = 1 : 100 : nlen
    RING2  =  [ RING1(ind:end,1); RING(1:ind-1,1)];   %%%% rotate and start the lattice at ind-th element
    [bd2, par2]=atx(RING2,0,1:length(RING2));
    emits2=par2.modemittance;
    demitsratio=(emits2-emits1)./emits1;
        
    results(ind, :) = [s00(ind), demitsratio, emits2];
    fprintf('%.4e ',results(ind, :));fprintf('\n');
end
plot(results(:,1),results(:,2),'b-');
hold on; 
plot(results(:,1),results(:,3),'r-');
legend('dex/ex0','dey/ey0');

save('./esrf_emits.txt','results','-ascii');

z-duan, 2019-05-30:
[test_emit_esrf_1.txt](https://github.com/atcollab/at/files/3236334/test_emit_esrf_1.txt)

z-duan, 2019-06-02:
Sorry, I found out the numerical difference stepsize need to be set properly, since the default value 6.055454452393343e-006 could be too large for the concerned calculation, for example
global NUMDIFPARAMS;
NUMDIFPARAMS.XYStep =  1e-8;
NUMDIFPARAMS.DPStep =  1e-8;

closed 2019-06-02 by z-duan

lfarv, 2019-06-02:
@z-duan : Thanks for giving the solution so rapidly! According to your findings, I think we should reduce the default step size to avoid falling again in that trap. Furthermore, I could not find any justification for the very strange default values. I'll prepare a pull request to clarify this.

ZeusMarti, 2019-06-13:
I encountered a similar problem with atx on our lattice (attached):

```
load('ring_atx_wrong.mat')
[lo,pa]=atx(ring,0);
 pa.modemittance
```

gives 7.8269e-06, while:

```
ats=atsummary(ring,'NoDisplay');
ats.naturalEmittance
```

gives 4.5276e-09, which is much more reasonable. As @z-duan said it can be solved by just:

```
global NUMDIFPARAMS;
NUMDIFPARAMS.XYStep = 1e-8;
NUMDIFPARAMS.DPStep = 1e-8;
```

I wonder if maybe the default should be 1e-8?

[ring_atx_wrong.txt](https://github.com/atcollab/at/files/3284866/ring_atx_wrong.txt)
Zeus