atcollab/at #118: pyat: Radiation/Synchrotron Integrals

T-Nicholls, 2019-06-07:
I have been working on an interactive Linear Optics and lattice editing GUI for an accelerator physicist here at Diamond.
> Side note: a big thank you to @lfarv as the `at.plot` functionality has worked well and been very helpful.

He would like some Global Linear Optics Properties displayed which are not currently calculated in pyat (Linear Dispersion Action, Momentum Spread and the Damping Partition Numbers). As I understand it, these are all calculated from / related to the Radiation Integrals. To save time we are working around the issue and trying to calculate them from the outputs of functions currently in pyat. However, after discussion with him he has said that the accelerator physics group at Diamond are very interested in the Synchrotron Integrals when developing lattices. So, although not that important it would be nice if we could calculate them in pyat.
- Has anyone else encountered this need?
- If yes, did you write any Python code for calculating them?
- If not, would anyone else be interested if I wrote it?
- If I were to go about writing a Python function for this, would the best method just be to convert the current Matlab function?

lfarv, 2019-06-07:
Short answer since I’m not at home and I have no access to the MATLAB code: the momentum spread and damping partition numbers can be directly derived form ´ohmienvelope’ results (see atx.m). They are valid for any lattice, without needing the radiation integrals. These radiation integrals are defined only for uncoupled lattices. The computation from the optical functions at the ends of elements is straightforward (though formulae are quite long) and I think this is done in ´ringparam.m’. More details when I’m back in 10 days. 
>

T-Nicholls, 2019-06-08:
@lfarv, Thank you for the quick reply I will take a look at `atx.m` on Monday.

T-Nicholls, 2019-06-21:
@lfarv, thank you for the direction, the methods found in `at.x` worked a treat.
My first attempt at a Python synchrotron integrals function can be found on #120, but requires more work.

T-Nicholls, 2019-07-06:
Closing as this has now been nicely implemented by @lfarv on #122

closed 2019-07-06 by T-Nicholls