atcollab/at #126: pyat: Update numpy version in requirements.txt

T-Nicholls, 2019-08-19:
The latest update to pyat file loading introduced this line, to `load/matfile.py`:
```
numpy.set_printoptions(linewidth=200, floatmode='unique')
```
`floatmode` was added to `arrayprint` in 1.14, and so older numpy versions will fail when trying to import at. We currently specify `numpy>=1.10` in our `requirements.txt`, so perhaps whoever merges the next pull request could tack this change onto the end.
@willrogers @lfarv

lfarv, 2019-08-19:
`floatmode='unique'` is set so that no precision is lost in the text representation of any number. As a first fix, I propose to:
1. move the `set_printoptions` inside the `save_m` function (as it is done for `save_repr`).
2. put the `set_printoptions` in a `try: ... except:` clause and warn that the representation may cause a precision loss if it fails.

Then one could look at the best options for numpy < 1.14 so that the representation is accurate enough. However I cannot test that easily…

T-Nicholls, 2019-08-19:
I think consistency with `save_repr` and a try-except are good. As for numpy versions < 1.14 I see 3 possible ways to proceed:

1. Scan the lattice to determine the maximum precision used and then set the `precision` in `set_printoptions` to that.
2. Decide on a reasonable default precision, and also let it be manually set via an argument to `save_mat` and `save_repr`.
3. Raise an error and require numpy>=1.14 for saving files.

Option 1 is inefficient, option 2 could result in data loss, and option 3 somewhat pins us to a version of numpy that is a year and a half old. Options 1&2 would also have the issue of excess 0's being printed, resulting in longer files.

lfarv, 2019-08-19:
Some good news:
- `.mat` files are binary, so they are not affected by the problem,
- for `.m` files, numpy arrays are converted to` list`s, so printing them is independent from the numpy version (`numpy.set_printoptions` is useless for those),
- I see a solution for `.repr` files using the `formatter` keyword, already existing in numpy 1.10.

So I 'll prepare a pull request to try and solve this. However, I would push for raising the requirements to numpy >= 1.14 is this is acceptable.

lfarv, 2019-08-19:
@T-Nicholls : Please have a look at #127 !

T-Nicholls, 2019-08-20:
Wow, that is a much more elegant solution then I envisaged.
I'd support raising the requirement to numpy>=1.14, however with the inclusion of #127 is there still a need?

T-Nicholls, 2019-08-20:
Actually, @willrogers is looking at loading Tracy files, so perhaps having `floatmode` available may be beneficial to him if he implements saving; if so, we should definitely raise the requirement.

willrogers, 2019-10-01:
I think this can be closed now. I have released accelerator-toolbox 0.0.3 which supports numpy 1.14.3 and above.

https://pypi.org/project/accelerator-toolbox/

closed 2019-10-01 by willrogers