atcollab/at #129: Significantly slower performance when looping through lattice

lcarver, 2019-09-05:
[lattice.zip](https://github.com/atcollab/at/files/3579400/lattice.zip)

Hello,

I have noticed a significant reduction in performance for some quite basic operations on the Lattice object. This reduction has been introduced at some point since the end of June. 

Please see below for a comparison between the master branch (up-to-date) and the errors_correction branch (not synched with master since end of June). I have attached a lattice called betamodel that can be used with the code below.

`import at, time`
`ring = at.load_mat('./betamodel.mat', key='betamodel')`
`ring = at.Lattice(ring) `
`bpmElems = ring.get_elements('BPM*')`
`start = time.time()`
`ind = ring.index(bpmElems[-1])`
`print(time.time() - start)`

If we compare the computation time (s) between the master and the errors_correction branch:
master:  2.9074273109436035 
errors_correction: 0.00014591217041015625

Cheers,
Lee

@swhite @carmignani

willrogers, 2019-09-11:
I expect @lfarv might have some ideas about this, but I think he is away at the moment. I don't immediately know what's happening here, but clearly 2.9 seconds is a long time.

lfarv, 2019-09-13:
`ind = ring.index(bpmElems[-1])` tests equality of the element `bpmElems[-1]` with all elements of `ring`. We recently introduced a new method `Element.__eq__` which implements that by building the element `repr` string. This is very slow compared to the previous behaviour comparing element `id`s.

I think we should revert to the previous behaviour, which is adequate for most cases (including @lcarver's search).

willrogers, 2019-09-19:
This is unfortunate, but I suppose I agree. The difference is that two different elements with the same parameters won't compare as equal, and I would prefer them to, but this performance problem is too big.

lfarv, 2019-09-23:
#131 reverts to the previous behaviour. In the future, one could create another `Element` method (`isequal` for instance) to compare element parameters, if necessary.

lcarver, 2019-09-30:
Thank you Laurent and Will for your work.

willrogers, 2019-09-30:
Can we close this issue?

lcarver, 2019-09-30:
Yes, I just checked on my side and the problem is no longer seen.

closed 2019-09-30 by lcarver