atcollab/at #133: IdTablePass

ZeusMarti, 2019-10-11:
Hi all,

I have a problem with the new implementation of IdTablePass.c. It was working well for me until [this](https://github.com/atcollab/at/commit/6fd8f7ceddfdad191c644d19ab348c8056560419#diff-7a33748c1c2660dcad345c0f90d14666) commit. This was on July 2018, drastic changes were made. I'm attaching an ID saved as a cell structure array that can be used with linepass [ID.txt](https://github.com/atcollab/at/files/3718274/ID.txt). Using the previous IdTablePass.c version:

>> linepass(ID,zeros(6,1))
ans =
     0
     0
     0
     0
     0
     0

Instead with the current version I get:

>> linepass(ID,zeros(6,1))
ans =
   1.0e-03 *
    0.0534
    0.1072
   -0.0160
   -0.0353
         0
    0.0000

which is not realistic... Has anyone had the same problem? If I'm not wrong, the commit was made by @lfarv, could you please take a look?

Many thanks,
Zeus

carmignani, 2019-10-15:
Hello @ZeusMarti,
I looked at the code and you are right, I think there is a bug in the new implementation.
In line 109 (and the same in line 158):
`xkick=atGetDoubleArraySz(ElemData,"xkick", &nx_map, &ny_map); check_error();` 
ny_map and nx_map are inverted, because the function atGetDoubleArraySz (the Matlab AT version of it) wants ny first and nx after as third and fourth inputs:
`static double* atGetDoubleArraySz(const mxArray *ElemData, const char *fieldname, int *msz, int *nsz)`
msz is number of rows (ny), nsz is number of columns (nx).

I can make a branch and a pull request, but I would like to be sure that the python version is correct too. 
@willrogers and @lfarv, is the function atGetDoubleArraySz for the pyat, in atelem.c line 169, also filling first number of rows then number of columns?

carmignani, 2019-10-15:
If you want a quick fix, to be able to do your simulations, you can just replace lines 109 and 158 with:
`        xkick=atGetDoubleArraySz(ElemData,"xkick", &ny_map, &nx_map); check_error();`
and then recompile.

ZeusMarti, 2019-10-15:
Thanks @carmignani, I already did my the simulations by replacing the file with the previous version. Now I'll be waiting for the problem to be fixed as everybody else. If you make the branch I'll be happy to try it. 
Ciao

carmignani, 2019-10-17:
Solved with pull request #134.

closed 2019-10-17 by carmignani