atcollab/at #137: lattice_pass modifies input particle when n_particles=1

lcarver, 2019-10-30:
I noticed some strange behaviour when using at.track.lattice_pass. Basically when I track a single particle, the function 'lattice_pass' modifies the input parameters 'rin'. For more than one particle this doesn't occur. I have attached an example ring, see below for the code used to create the error.

[betamodel.zip](https://github.com/atcollab/at/files/3788970/betamodel.zip)

```
import at
import numpy as np

ringMatFile = '/operation/beamdyn/matlab/optics/sr/S28F_restart/betamodel.mat'                  
latticeName = 'betamodel'  

ring = at.load_mat(ringMatFile, key=latticeName)     
ring = at.Lattice(ring)
bpms = np.array([ring.index(x) for x in ring.get_elements('BPM*')])  

rin = np.zeros((6,1)) 
res = at.track.lattice_pass(ring, rin, nturns=1, refpts=bpms)                        
print(rin)

[[ 1.19855818e-06]
 [-1.91101144e-09]
 [ 0.00000000e+00]
 [ 0.00000000e+00]
 [-4.21432351e-04]
 [-1.52451643e-05]]

```

However for n_particles > 1:
```
rin = np.zeros((6,2)) 
res = at.track.lattice_pass(ring, rin, nturns=1, refpts=bpms)                        
print(rin)

[[0. 0.]
 [0. 0.]
 [0. 0.]
 [0. 0.]
 [0. 0.]
 [0. 0.]]
```

While this is hardly critical (the tracking component still works well) I consider this to be a bug as no modification should occur on the input parameters of a function.

lcarver, 2019-10-31:
Thanks Laurent.

Could a flag be introduced in lattice_pass (and other tracking functions) that will provide the option to modify the initial conditions? I understand the need in point 1 where you want to minimize the number of copies as when you have a large number of particles this can become very slow as it doubles the memory being used. However I spend more time doing lots of single particle tracking rather than large arrays of particles so for me it would be much more convenient. Maybe a flag called copy_input where the default is True.

lfarv, 2019-10-31:
@lcarver : For the moment, can't you use:
```
...
rin = np.zeros((6,1)) 
res = at.track.lattice_pass(ring, rin.copy(), nturns=1, refpts=bpms) 
...                       
```
which will keep your `rin` unchanged ?

lfarv, 2019-10-31:
See #138 for a discussion on the best fix

lcarver, 2019-11-04:
Hi Laurent, I approved the changes in the pull request but I don't necessarily think it solves the issue that I raised. Maybe it's more of a philosophical question, but I don't think that `rin` should ever be modified when passing it to a function. You are correct in that there are easy workarounds (like using `rin.copy()` as you suggested) but maybe it's something that can be discussed in the future if you have strong opinions against it.  

Thanks, Lee

lfarv, 2019-11-11:
The bug is fixed in #138, so I remove the bug keyword, however the question raised by @lcarver is still open: should we provide another function similar to `lattice_pass` which does not modify its arguments ?

closed 2025-07-23 by swhite2401