atcollab/at #139: at.load_mat() imports PolynomA/B as array(dtype=uint8)

swhite2401, 2019-11-04:
Hello all,
I have noticed that at.load_mat() imports PolynomA/B as array(dtype=uint8) if a vector of integer is supplied by the matlab lattice consequently raising:
RuntimeError('The attribute PolynomA is not a double array.')
Since an array of integer is allowed in matlab (and I believe is the default in case of a vector of zeros) maybe the import could be forced with dtype=float for these elements?
What do you think?
Best regards,
Simon

lfarv, 2019-11-05:
@swhite2401 : can you post an example? I tried with various Matlab elements where I explicitly set PolynomB as int8in Matlab, and I always get an array(dtype=float64) in pyat…

lfarv, 2019-11-05:
I cannot reproduce the problem:

In Matlab:

```
>> mach=atringparam('test',6e9,1);
>> qp=atquadrupole('qf',1.0,0);
>> qp.PolynomA=uint8([0 0]);
>> qp.PolynomB=uint8([0 0]);
>> thmp=atthinmultipole('mp',int8([0 0]),int8([0 1]));
>> lattice={mach,qp,thmp};
>> class(lattice{2}.PolynomB)
ans =
    'uint8'
>> class(lattice{3}.PolynomB)
ans =
    'int8'
>> 
>> save testring.mat lattice
```

In python:

```
>>> ring=at.load_mat('testring.mat')
>>> print(ring[0])

Quadrupole:
	FamName : qf
	Length : 1.0
	PassMethod : QuadLinearPass
	NumIntSteps : 10
	MaxOrder : 1
	PolynomA : [0. 0.]
	PolynomB : [0. 0.]
	K : 0.0

>>> print(ring[1])

ThinMultipole:
	FamName : mp
	Length : 0.0
	PassMethod : ThinMPolePass
	MaxOrder : 1
	PolynomA : [0. 0.]
	PolynomB : [0. 1.]

```
All attributes are float… How do you get uint8's ?

swhite2401, 2019-11-05:
@lfarv here is the example that was posing problem, attached the TL2 file used in the example. You can see PolynomA&B are vectors of integers.

Lattice file:
[betamodel.zip](https://github.com/atcollab/at/files/3810012/betamodel.zip)

Sample code:
```
>>> LatticeFile='betamodel.mat'
>>> LatticeVar='betamodel'
>>> line = at.load_mat(LatticeFile,key=LatticeVar,energy=6.0e9,periodicity=1) 
>>> print(ring[2])
Corrector:
	FamName : CV0
	Length : 0.25
	PassMethod : StrMPoleSymplectic4Pass
	KickAngle : [0. 0.]
	PolynomA : [0 0 0 0 0]
	PolynomB : [0 0 0 0 0]
	MaxOrder : 4
	NumIntSteps : 20
	Device : tl2\ps-c1\cv0
```

swhite2401, 2019-11-05:
@lfarv And then for the tracking:

```
>>> import numpy as np
>>> part = np.zeros(6)
>>> at.track.lattice_pass(line,part,nturns=1,refpts=1)
Error in tracking element 2.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/operation/beamdyn/at/pyat/at/tracking/track.py", line 58, in lattice_pass
    return atpass(lattice, r_in, nturns, refs, int(keep_lattice))
RuntimeError: The attribute PolynomA is not a double array.
```

lfarv, 2019-11-05:
Ok, I got it! The problem is that `PolynomA/B` is not expected in a corrector, so this attribute does not go through the normal type coercion… I'll look into that.

swhite2401, 2019-11-05:
@lfarv Ok great, thanks!!

lfarv, 2019-11-05:
Fixed in #140.
@swhite2401 : can you try it?

swhite2401, 2019-11-06:
@lfarv Works for me! Perfect thanks a lot.
Simon

lfarv, 2019-11-11:
Fixed in #140

closed 2019-11-11 by lfarv