atcollab/at #148: BndMPoleSymplectic4Pass strange behaviour

ZeusMarti, 2020-02-18:
Hi all,

I recently noticed that BndMPoleSymplectic4Pass fringe or edge in AT2.0 gives largely different energy acceptance results compared to AT 1.3 for the ALBA lattice (combined function bending magnets).

It could be I'm not defining the ALBA dipole appropriately in AT2.0. In case someone wants to help with it I'm adding some more details here. The following code tracks through a half dipole of ALBA:

BEND.FamName='BEND';
BEND.Length=0.6869;
BEND.MaxOrder=3;
BEND.NumIntSteps=10;
BEND.BendingAngle=0.0982;
BEND.EntranceAngle=0.1035;
BEND.ExitAngle=0;
BEND.ByError=0;
BEND.K=-0.5669;
BEND.R1=eye(6);
BEND.R2=eye(6);
BEND.T1=[0 0 0 0 0 0];
BEND.T2=[0 0 0 0 0 0];
BEND.PolynomA=[0 0 0 0];
BEND.PolynomB=[0 -0.5669 -0.2211 0];
BEND.PassMethod='BndMPoleSymplectic4Pass';
BEND.FringeInt1=0.7000;
BEND.FringeInt2=0;
BEND.FullGap=0.0360;
BEND.Energy=2.9995e+09;
LINE={BEND};
small=0.1e-3;
Rin=eye(6)*small+[zeros(4,6);1e-2*ones(1,6);zeros(1,6)]; %with energy offset!
T=(ringpass(LINE,Rin)-Rin)/small;

Between AT20 and AT1.3 I get the following difference

T20-T1.3= 1.0e-03 * [0.1043         0    0.0000         0         0         0
                      0.1661         0    0.0000         0         0         0
                      0         0   -0.0922         0         0         0
                      0         0   -0.1234         0         0         0
                      0         0         0         0         0         0
                      0.0051         0    0.0000         0         0         0]

This difference disappears if I remove the energy offset ("Rin=eye(6)*small"). Also, the difference does not depend on the starting angle if the initial position is zero in both planes. This seems to indicate that there is an energy dependent effect at the dipole entrance, so I guess the new fringe or the edge part of the code are playing a role.

Thanks,

ZeusMarti, 2020-03-06:
Hi,

I think it is due to a typo on "atphyslib.c" that only is active when using the legacy method. In lines 41 and 73, it reads:

fy = inv_rho*tan(edge_angle-fringecorr/(1+r[4]));

While it should read:

fy = inv_rho*tan(edge_angle-fringecorr)/(1+r[4]);

Do you agree?

Cheers,

carmignani, 2020-03-06:
Hello Zeus,
looking at the history of the file, I see that I wrote this piece of code in 2017! We had a few meetings at ESRF with @lnadolski to define the three methods and the idea was to keep compatibility with the old version for method 1 and just add two different methods (2 and 3). 

I looked at the lines 41 and 73 as you suggested, but I think they are correct. It looks to me exactly what it was done in the function edge_fringe in at 1.3, so I don't think it is a typo. In the image here you can see the old version of the code.

[image]

ZeusMarti, 2020-03-06:
Hi Nicola,

Thanks for answering! Yes, sorry I was confused. In fact I just checked that our AT1.3 version has implemented the SOLEIL method of AT2.0 (I know remember talking with Laurent about it many years ago). Then, I guess that the discrepancies (**a factor 4 in the lifetime calculation**) that I observe may be due to that.

I see that in BndMPoleSymplectic4Pass.c (AT2.0) the call to the fringe is the following:

edge_fringe_entrance(r6, irho, entrance_angle, fint1, gap, FringeBendEntrance);

So in the "edge_fringe_entrance", the selected method is 1 (legacy). Is there a way to change it to the SOLEIL method without rewriting BndMPoleSymplectic4Pass?

Cheers,

carmignani, 2020-03-06:
Hello,
yes, there is a way to use the Soleil method, which is the number 2. You have to add the fields "FringeBendEntrance" and "FringeBendExit" to the element and the value of the field must be 2 if you want the Soleil version and 3 if you want the ThomX version. 
In case you don't have those two fields, the default method (number 1) is used.

It's interesting that you see a factor 4 difference in lifetime! I didn't think it would be a so huge effect.

ZeusMarti, 2020-03-06:
True, I had not noticed that in our standard lattice but on some more exotic cases the tracking start to diverge noticeably after turn 100 or so. The divergence happens in the vertical plane and there it hits the chamber. This is also consistent with the fringe difference which affects the vertical plane only.

ZeusMarti, 2020-03-06:
By the way, thanks a lot Nicola!

ZeusMarti, 2020-03-11:
Just for the recod, I'm attaching a picture of it. I must insist in the fact that we have observed this just for an specific optics and sextupole settings, it is generally not like that. In this case the energy deviation is 0.02 and the initial coordinates 0.01 mm in both planes.

[image]