atcollab/at #164: closed orbit in pyat ohmi envelop

swhite2401, 2020-07-10:
Hello @willrogers @lfarv @lcarver ,

I was working on quatum diffusion and saw a potential issue in ohmi_envelop

I the present code there is this line:
`    bb = [find_mpole_raddiff_matrix(elem, orbit, energy)
          if elem.PassMethod.endswith('RadPass') else b0 for elem in ring]`

while I would have expected

`    bb = [find_mpole_raddiff_matrix(elem, orbit[i], energy)
          if elem.PassMethod.endswith('RadPass') else b0 for i,elem in enumerate(ring)]`

to properly take into account the closed orbit at elem, it seems to me that in the present implementation 
the initial orbit is taken for all elements.

Cheers,.
Simon

swhite2401, 2020-07-10:
In matlab at it is implemented as follows

`B(radindex)=cellfun(@findmpoleraddiffmatrix,...
    ring(radindex),orb(radindex),'UniformOutput',false);`

I suppose this confirms the problem, I will correct it in the branch quantdiff so
that the bugfix is implemented together with the addition of quantum diffusion

lfarv, 2020-07-11:
@swhite2401 : I agree, it's a bug. Thanks !

closed 2020-08-12 by swhite2401