atcollab/at #168: Calculation of synchrotron radiation integrals not consistent

TeresiaOlsson, 2020-07-24:
The calculation of synchrotron radiation integrals doesn't seem to be done consistently in all functions resulting in different output depending on which function you use. Especially output like energy loss etc is no longer correct when including IDs using the GWigSymplecticRadPass function.

I suggest to implement a function that calculates the radiation integrals correctly and then call that function from all other functions that use the radiation integrals.

carmignani, 2020-07-24:
I think ringpara computes correctly the radiation integrals, at least without wigglers. But I agree, we can write a function just for the radiation integrals and then use it everywhere they have to be computed.
What functions did you use for the radiation integrals?

T-Nicholls, 2020-07-24:
Hi @TeresiaOlsson,
Have you taken a look at #121 yet, it might address some of the problems you mention? It incorporates some work I did last year to try and get the updated (and hopefully numerically correct) `GWigSymplecticPass` and `GWigSymplecticRadPass` methods working with AT2.0 and pyAT. The last I heard the changes had been passed to the Diamond AP group for testing, but I am unsure if they concluded whether the functions were accurate or not; @willrogers may have a better idea of their status.

TeresiaOlsson, 2020-07-24:
Hi,

I'm in the Diamond AP group and we are testing it now. So far it seems like the passmethods are working correctly, but other functions need updating to output correct numbers when including them in the lattice.

We tried atsummary, atx and ringpara and none seem to include the effect of the wigglers on the global parameters. The cause seems to be that the calculation of the radiation integrals doesn't include it. There are also other functions like atsetRFCavity, atenergy etc that needs updating to work correctly with wigglers.

Then we will take a look at it and propose changes.

lnadolski, 2020-07-24:
dear @TeresiaOlsson 

Can you send us an full example for helping us to debug ?

Laurent.

T-Nicholls, 2020-07-24:
I am glad to hear that it is being worked on and that the pass methods are correct.
I definitely agree that all the functions should be consistent and centralising the calculation of the radiation integrals would be sensible.
I am not surprised that `atsummary`, `atx`, and the other functions do not provide the correct numbers as they were not updated for wigglers. I am, however, concerned that `ringpara` doesn't function correctly, as it was updated to use `RadIntegrals.m` to calculate the added effects of the wigglers on the radiation integrals.

TeresiaOlsson, 2020-07-24:
Could be that I'm wrong and ringpara is working better than the other functions. I'll have a look at RadIntegrals.m and see if that could be used to solve the the problems with the other functions.

We will prepare an example.

marlibgin, 2020-07-24:
Hi guys, I have prepared an example with one ID in a Diamond-II lattice. It's a SCW working at 4.2T so the effect should be quite evident,  I am trying to upload the *.m but does not seem to support the type. How can I share it with you?

Marco

lnadolski, 2020-07-24:
I do not knwo why you cannot download the file.
You can email me directly nadolski@synchrotron-soleil.fr

T-Nicholls, 2020-07-24:
@marlibgin, for future reference, you should be able to attach the file here if you change the extension to `.txt` by renaming the file.

marlibgin, 2020-07-24:
> @marlibgin, for future reference, you should be able to attach the file here if you change the extension to `.txt` by renaming the file.
Hi @T-Nicholls , yes ... I thought it after, but wasn't sure about it, thanks for clarifying. 

Marco

[AT2_M_H6BA_15_1_1_ID12.txt](https://github.com/atcollab/at/files/4974634/AT2_M_H6BA_15_1_1_ID12.txt)

lfarv, 2020-07-25:
I'm sorry to see that at the moment no function in AT computes properly the radiation integrals.

First, `atx` (based on `ohmienvelope`) cannot be compared with the other functions: `ohmienvelope` neither computes nor uses any radiation integral. It computes **equilibrium emittances** and **damping times**. This has advantages:

- it can be used with H/V coupling,
- it can take into account radiation from quadrupoles, sextupoles…
- it is computed with cavities and radiation ON.

And a drawback:

- it needs significant work to include wigglers: it requires the computation of the diffusion matrix. At the moment, it is computed for any element with a constant field represented by a PolynomA/B (not the case of wigglers). To include wigglers, `findmpoleraddiffmatrix` must be modified.

On the other side, `atsummary` and `ringpara` compute radiation integrals. These are based on beta and dispersion functions, so keep in mind that this needs:

- no H/V coupling
- **NO CAVITY** !
- **NO RADIATION** !

There is no check for these conditions, so you have to be careful ! I know that `atlinopt` gives numbers with cavities and radiation, but you should not rely on those.

In addition, both functions (including the modified version of `ringpara` in #121) are imprecise because the radiation from dipoles is computed by splitting the dipole in slices instead of using the analytical integral (see the python version). Depending on the way you describe the bending magnets (explicitly splitted or not), you get different results.

So I agree with the idea of factorising the computation of radiation integrals in a single function, but using the analytical integral of I5 in dipoles. And for a more general computation, one might look at the computation of the diffusion matrix to allow wigglers in `ohmienvelope`.

TeresiaOlsson, 2020-07-26:
@lfarv Do I understand you correctly that the radiation integrals are calculated in a better way in pyAT? It was my feeling that it might be the case, but I wasn't sure.

I also think it's a good idea to make a function that calculates the analytic integral. Not at least because it's a very good tool for quickly benchmarking what happens when you put in IDs against other codes such as Elegant. However, eventually we would also like to do collective effects simulations including the effects of the IDs and that requires `atfastring.` Would that require the changes you mention to the diffusion matrix?

lfarv, 2020-07-27:
@TeresiaOlsson: Yes, pyAT is (slightly) better since the numerical integration in the dipoles is replaced by the analytical integration. But IDs are not yet included. And at the moment it suffers from bug #164 appearing for non-zero closed orbit (corrected soon).
For integration in wigglers, I am not aware of any analytical integration, so we have to use the numerical integration as done in `RadIntegrals.m` (I did not check it yet). Both in Matlab and Python!

On the other hand, `atfastring` is based on the same formalism as `ohmienvelope`: it does not compute or use radiation integrals. So you are right, taking wigglers into account needs modifying `findmpoleraddiffmatrix`. Or more exactly, writing a new "`findwigglerraddiffmatrix.c`" function.

lfarv, 2020-08-28:
#178 is an attempt to clarify the computation of radiation integrals. However, it looks like the existing treatment of wigglers is wrong (see the discussion). Did anybody ever checked the contribution of a wiggler? If so, I would like to get the wiggler description and the expected and obtained results.

carmignani, 2021-01-11:
Hello @lfarv,
I think there is a bug in DipoleRadiation.m. The function looks for the field K of the dipoles, but K is not a mandatory field.
Now ringpara fails with the ESRF lattice, because the K value is not found in the dipoles.
The old ringpara was looking at both K and polynomB(2), if I remember well, and then it was giving a warning in case the two were different.