atcollab/at #169: Periodicity not accounted for properly in all function

swhite2401, 2020-08-11:
@lcarver , @lfarv the periodicity parameter is not accounted for in some cases leading to different results in matlab AT and pyAT for the detuning of the tune from tbt data.
For now only Periodicity=1 gives consistent results

swhite2401, 2020-09-04:
Hello @lfarv, comparison with matlab becomes tricky because of this periodicity issue. I am not very familiar with this parameter, which paramters are affected? Tunes and Qp? Something else?

lfarv, 2020-09-04:
@swhite2401 : the `periodicity` should affect as many parameters as necessary: when a ring is defined as "description of one superperiod" + "number of periods", **all the results should be given for the whole ring**. Circumference, tunes, radiation integrals, chromaticity, revolution time, tune shifts, energy loss per turn… Everything !
To get the results for 1 super period, just turn periodicity to one, easy !
There are probably still missed parameters in Matlab, but that's the idea.

swhite2401, 2020-09-04:
Ok I'll fix the ones I know of then!

swhite2401, 2020-09-04:
It seems trickier than I thought, as far as I can see only ring_parameters.py is taking it into account properly.
Even in matlab it is incoherent, for instance atfastring take the full ring detuning but only partial chromaticity

TeresiaOlsson, 2020-09-04:
I discovered yesterday that this problem also exist in atenergy in matlab. When you on purpose want to use 1 super period, the function discovers that you don't have a full circle, calculates how many periods you need and gives you the energy loss of the full ring. Then output from all functions that uses the energy loss from atenergy becomes wrong and you might not realise it unless you know what to expect.

swhite2401, 2020-09-04:
Ok, so for now I will fix fastring.py such that it behaves as the matlab version and leave this issue open as a place to report the function that are not behaving properly

TeresiaOlsson, 2020-09-08:
I have discovered the same problem with atx, but the cause seems to again be atenergy. It results in atx giving different results than atsummary and ringpara.