atcollab/at #171: get_twiss() usefull?

swhite2401, 2020-08-11:
Hello @lfarv , @willrogers ,

I took a look at linear.py and it seems to me that get_twiss() does not have any added value as it does exactly the same as linopt with coupled=False.
Is there any good reason to keep it?

If this function is required, it could at least be 

`def get_twiss(ring):
    return linopt(ring, coupled=False)`

to remove unnecessary duplicated code. What do you think?

T-Nicholls, 2020-08-11:
Hi @swhite2401,

They are almost identical however, unlike `linopt`, `get_twiss` does not calculate `A`, `B`, `C`, or `gamma` and, as a result, it is 1.5-2x faster than uncoupled `linopt`. Though the run time difference isn't as large as it used to be (#90), `get_twiss` may still useful for high-frequency recalculations of twiss data where `A`, `B`, `C`, and `gamma` are not needed.

willrogers, 2020-08-12:
Thanks @T-Nicholls, I remember now. As @swhite2401 didn't realise this (perfectly reasonably) then it would be worth adding a comment to the `get_twiss()` function stating this explicitly.

swhite2401, 2020-08-12:
Hi @T-Nicholls,

I just checked and get_twiss is faster by 30%. The 2 lines that make the difference are:

`        lindata['A'] = [ms.dot(A.dot(_jmt.dot(ms.T.dot(_jmt.T))))
                        for ms in msa]
        lindata['B'] = [ms.dot(B.dot(_jmt.dot(ms.T.dot(_jmt.T))))`

Duplicating a 100 lines of code is not efficient and maintenance gets more complicated. Also
AT users are used to work with linopt and I think get_twiss() will simply never be used.

Having a faster minimal twiss computation could however be useful in some cases like matching
(not implemented yet in pyAT) so I suggest that an option is added as input argument in linopt
to return or not A and B and get rid of get_twiss()

This will improve readability and future maintenance/devlopments.

swhite2401, 2020-08-12:
@willrogers @T-Nicholls 
I have started a branch to adapt linopt to transfer lines, following this discussion I also propose an implementation that matches the speed of get_twiss() for coupled = False by returning NaNs for A,B,C,g

This would allow to have a single function the compute twiss functions

You can take a look at the branch twissline

willrogers, 2020-08-14:
Yes, I think this is appropriate, as long as we are clear that the speeds are equivalent.

The speed of calculation matters for our Python 'virtual accelerator' where it affects the update rate we can maintain. I would test this but I have a LOT of different things on at work at the moment.

swhite2401, 2020-08-14:
Ok great, Thanks. I went on with the implementation of the transfer line which is now ready for test and submitted to Laurent for review so you will see significant changes.
See pull request #172 
get_twiss() was removed in that version but  I can bring it back for comparison purposes, let me know

willrogers, 2020-09-02:
@swhite2401 thanks for that, I've just checked and it looks like `get_twiss` is indeed redundant. The times are the same for the same operations and from a quick inspection the results appear to be identical.

Go ahead and remove it again.

swhite2401, 2020-09-02:
Thanks @willrogers , I close this issue

closed 2020-09-02 by swhite2401