atcollab/at #195: decodeatargs bug?

ZeusMarti, 2020-11-26:
Hi all,

On line 21-25 the decodeatargs code reads:

nopass    = @(arg) ischar(arg) && ~isempty(arg) && isempty(regexp(arg,'.*Pass$','once'));
% get index of first optional char Argument
chararg   = find(cellfun(nopass,[args {'x'}]),1); 
% get all optional argument (ressources)
rsrc      = args(chararg:end);

I get an error at some elements (wigglers), which is solved if I do the following modification:

nopass    = @(arg) ischar(arg) && ~isempty(arg) && ~isempty(regexp(arg,'.*Pass$','once'));
% get index of first optional char Argument
chararg   = find(cellfun(nopass,[args {'x'}]),1); 
% get all optional argument (ressources)
rsrc      = args((chararg+1):end);

Does it seems right to you too?

Cheers,

lfarv, 2020-11-28:
Hello @ZeusMarti,
Can you give an example of element definition which gives an error ?
The function looks for the 1st "non-passmethod" string in the argument list, which should be an option name, while your version would skip this option **name** and point  to the following option **value**…

ZeusMarti, 2020-12-02:
My mistake, my modification was solving an issue but not in the way I thought.

lfarv, 2020-12-02:
@ZeusMarti : then may I close the issue ?

closed 2020-12-02 by ZeusMarti