atcollab/at #197: Violation of Naff license

matsievskiysv, 2020-11-28:
This repo contains a number of licenses including [Naff](https://raw.githubusercontent.com/atcollab/at/master/doc/licenseFiles/NaffLicense.txt), which has the following line:
> !**************************************************************************
!  THIS PROGRAMM  CANNOT BE COPIED, 
!  DISTRIBUTED NOR MODIFIED WITHOUT THE AGREEMENT OF THE AUTHOR.
!**************************************************************************

I guess that this makes this repo illegal.
I, for one, forked and modified code without asking anybody.

lfarv, 2020-11-29:
@lnadolski: Can you comment on the possibility to create forks of AT?

lnadolski, 2020-11-30:
This header is indeed from legacy.
This is fine to use NAFF in the context of accelerators.
Forks as allowed.

closed 2020-11-30 by lnadolski

matsievskiysv, 2020-11-30:
Then remove this phrase maybe?