atcollab/at #204: Jupyter tests

matsievskiysv, 2020-12-04:
*This is more of a PSA then an issue*
As of  #198 it's possible to run AT on [jupyter](https://jupyter.org/) server. Self documented jupyter notebooks are well suited for AT projects; local installation of AT is not required.

I've created a docker container of Jupyter server so you can effortlessly try it out.

In order to start a server:

1. Install [Docker](https://docs.docker.com/get-docker/)
1. Run command `docker run -d --name at -p 8888:8888 matsievskiysv/at`. It will pull server image and start it
1. Run `docker logs at --follow` and write down connection token (will appear on screen as `http://127.0.0.1:8888/?token=<token>`)
1. From web-browser go to `http://127.0.0.1:8888/` page and insert token there
1. Try out read-only examples from `demos` folder or create a new notebook
1. Report any issues

lfarv, 2020-12-07:
Hello @matsievskiysv. I am quite successful with your docker image. It is very convenient for tests, and I appreciate the layout of notebooks you created from the demos.
At the moment, I just have a problem with `atplot`:

```
a=load(fullfile(atroot,'../pyat/test_matlab/hmba.mat'));
mring=a.RING;
atplot(mring);

```
returns:
```
p4 = -15.453
Inline plot failed, consider trying another graphics toolkit
error: r(1): out of bound 0 (dimensions are 0x0)
error: called from
    _make_figures>safe_print at line 125 column 7
    _make_figures at line 49 column 13
```
This is specific to the this lattice: with the other test lattice
```
a=load(fullfile(atroot,'../pyat/test_matlab/dba.mat'));
```
`atplot` works.

matsievskiysv, 2020-12-08:
@lfarv should be fixed in https://github.com/atcollab/at/commit/feee5c04d02d5857086736c863f7731a597e54a7
You'll need to update your image now. For this:
1. `docker stop at` - stop jupyter container
1. `docker rm at` - remove jupyter container
1. `docker pull matsievskiysv/at` - pull new version of the image
1. Start new jupyter container as described [above](https://github.com/atcollab/at/issues/204#issue-757184242)

**Note:** this will remove everything in your jupyter server. Save your notebooks before updating image

lfarv, 2020-12-08:
Hi @matsievskiysv: sorry for my naive questions:

1. Though I am mounting local volumes to run the `at` container, I do not know how to copy the notebooks (in the container) to my volume (or back, after upgrade): I do not know the file path of the notebook root! Or could I store the notebooks on my local volume?
2. I understood Docker uses split images so that some pieces are reusable. But it looks like I need to reload the whole thing when only `at` is updated. 
3. By changing the octave search path, I can run `at` from my local copy (as long as I do not need to recompile anything), but is there another way to create/edit functions directly?

matsievskiysv, 2020-12-09:
> Though I am mounting local volumes to run the at container

No, this has to be done explicitly (described below). Commands  is the [comment](https://github.com/atcollab/at/issues/204#issue-757184242) above start completely isolated container.

> I do not know how to copy the notebooks

For single file you may just use Upload/Download buttons
[image]

> I understood Docker uses split images so that some pieces are reusable.

Yes. This is why when you `docker pull`, it only downloads layers that were changed.

> I need to reload the whole thing when only at is updated

Because this image is more for at users, not developers (see below for alternative workflow)

> is there another way to create/edit functions directly

You may just enter functions directly in the notebook. Newly entered function will override the existing one.

# Mounting local folder

Mounting local folder lifts some limitations described above. It will allow container and local file system to use the same folder.
Contents of this folder will be preserved between image changes, so one would not need to save all notebook manually.

But some preparation is needed.
1. Stop running `at` containers
   ```bash
   > docker stop at
   > docker rm at
   ````
1. Create local folder and set permissions
   ```bash
   > mkdir $HOME/atjupyter
   > chmod 777 $HOME/atjupyter
   ````
   You may use any folder name/location you want.
1. Start `at` container
   ```bash
   > docker run -d --name at -p 8888:8888 -v "$HOME/atjupyter:/home/jovyan/work" matsievskiysv/at
   ```

This is enough if you just want to have persistent notebook storage.
For using a local copy of `at`
1. Clone `at` Octave branch to mounted folder
   ```bash
   > git clone https://github.com/atcollab/at.git --depth 1 -b octave_compatibilty $HOME/atjupyter/at
   ````
1. Create a new Octave notebook in `work` directory and from there issue commands:
   1. ```octave
       !rm ~/.octaverc
       ```
       remove bundled `at` version from path 
   1. Click `Kernel->Restart` -- clear `path` variable
   1. ```octave
       !find ~ -iname "*.mex" -delete
       cd ~/work/at/atoctave
       bootstrap
       savepath
       ```
       Recompile `mex` files and use local copy of `at` in the future

After these steps Octave will use `at` version cloned into `$HOME/atjupyter/at`

lfarv, 2020-12-09:
Thanks for your help. Your solution is close to what I did: I was mounting my local AT repository as `/at` on the container. But then I had to rebuild the Octave path for each new session. Your solution looks much better: by replacing /home/jovyan/work with mine, hopefully I can save notebooks locally

matsievskiysv, 2021-01-21:
I temporary disabled jupyter authentication to simplify testing.

matsievskiysv, 2021-01-30:
Major update.
Now `docker-compose` project starts a multiuser JupyterHub server with enabled TLS encryption.
It may be used "as is" for small classes and departments.

JupyterLab is not yet supported.

matsievskiysv, 2021-01-31:
I was able to add support for JupyterLab. It's now a default notebook environment.

closed 2021-01-31 by matsievskiysv