atcollab/at #218: Make pyat-0.1.0 release using Github Actions

willrogers, 2021-01-20:
Previous releases were made using Travis CI, but we are now using Github Actions for CI.

Useful links:

* [cibuildwheel docs](https://cibuildwheel.readthedocs.io/en/stable/)
* [Example Gtihub Action for deploying to PyPI](https://github.com/joerick/cibuildwheel/blob/master/examples/github-deploy.yml)
* [Accelerator Toolbox on PyPI](https://pypi.org/project/accelerator-toolbox/)

It is a tricky process, but is mostly working now. Currently failing:

* pypy on MacOS. This is present on the 0.0.4 release.

willrogers, 2021-01-20:
@lfarv at the bottom of `at.c` I see:

```
#if PY_MAJOR_VERSION < 3
int main(int argc, char *argv[]) {
    /* Pass argv[0] to the Python interpreter */
    Py_SetProgramName(argv[0]);

    /* Initialize the Python interpreter.  Required. */
    Py_Initialize();

    /* Add a static module */
    initatpass();
    return 0;
}
#endif
```

As far as I can tell `Py_Initialize()` and `Py_SetProgramName()` are not required, because they are specific to embedding a Python interpreter; removing them makes it possible to build wheels for PyPy, not that that is particularly important.

Do you remember more? It is a while since I looked at this code. Do we need the main function at all?

willrogers, 2021-01-20:
If I remove that `main()` function all the tests pass on Windows, Mac and Linux.

lfarv, 2021-01-21:
Perfect! As far as I remember, that block of code came from a tutorial on writing extensions, but I agree that there is no need for a `main()` here. If everything works without it, go on!

willrogers, 2021-01-21:
Is there anything particular to add before I make a 0.1.0 pyat release?

@lfarv @swhite2401

lfarv, 2021-01-21:
@willrogers, @swhite2401 : I would just merge #219 (bug fix), otherwise it looks like a good time to push a 0.1.0 release. Any objection ?

willrogers, 2021-01-25:
Released here:

https://pypi.org/project/accelerator-toolbox/

closed 2021-01-25 by willrogers