atcollab/at #232: Matlab R2017a and R2015a back compatibility

ZeusMarti, 2021-03-02:
Hi,

I just updated my local repo and got problems with R2015a and R2017a. After atmexall I get this error:

Error using atpass
Element # 0: Library file: /home/zeus/controlroom/at2_2021_test/atintegrators/IdentityPass.mexa64, No passFunction or
trackFunction available

The corresponding file is there, but for some reason atpass does want to use it...

With R2015a more errors appear related to the broadcasting in array operations introduced in R2016a. The errors could be easily removed by using "repmat" in the recent modifications of findm44, findm66, findorbit4 and findorbit6.

With R2019a I have no problems. Don't we want to keep older versions compatibility?

Cheers,

lfarv, 2021-03-04:
The `Install.md` document mentions that AT is presently compatible with releases 2016b and above. The reason is that R2016b introduced a modification on the the way Matlab handles element-wise operations on matrices:
- before R2016b, an operation (+ for instance) was possible if the dimensions of matrices were **identical**
- now, is requires that the dimensions are "**compatible**". It means that along a given axis, the dimensions are either identical or one is equal to 1. In this case the data are copied along that dimension to match the dimension of the other matrix. 

Since that time, this new feature has been used (even without noticing), so that now it is necessary.
This being clearly identified, it should be possible to restore the compatibility with previous releases, though it will be difficult to ensure that no instruction is forgotten. But is it worth the effort?

Concerning 2017a, it should work. It looks like the mex file is wrong (no entry point). Can you try to remove it, ensure that Matlab is not picking a wrong one somewhere else (now it should be missing) and rebuild? If the problem persists, can you post the output of `atmexall`?

ZeusMarti, 2021-03-04:
Sorry, I had not read the "Install.md" recently. R2016b or avobe OK. Looking at atmexall.m there is parts of the code that act diferently even for Matlab version R2014, I guess this is just old unharmful code.

Regarding R2027a, removing all the mex files I get:

```
Error using atpass
Element #0: PassMethod 'IdentityPass' is not on MATLAB search path
```
Running atmexall again I get the former error message:

```
Error using atpass
Element # 0: Library file: /home/zeus/controlroom/at2_2021_test/atintegrators/IdentityPass.mexa64, No passFunction or
trackFunction available
```

The output of atmexall is the following:

```
mex -DGLNXA64 -largeArrayDims -silent -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators -ldl -outdir /home/zeus/controlroom/at2_2021_test/atmat/attrack /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.c
mex -DGLNXA64 -largeArrayDims -silent -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators -outdir /home/zeus/controlroom/at2_2021_test/atmat/atphysics/Radiation /home/zeus/controlroom/at2_2021_test/atmat/atphysics/Radiation/findmpoleraddiffmatrix.c
mex -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atmat/atphysics/NonLinearDynamics /home/zeus/controlroom/at2_2021_test/atmat/atphysics/NonLinearDynamics/RDTelegantAT.cpp
mex -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atmat/atphysics/nafflib /home/zeus/controlroom/at2_2021_test/atmat/atphysics/nafflib/nafflib.c /home/zeus/controlroom/at2_2021_test/atmat/atphysics/nafflib/modnaff.c /home/zeus/controlroom/at2_2021_test/atmat/atphysics/nafflib/complexe.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BendLinearPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndMPoleSymplectic4E2Pass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndMPoleSymplectic4E2RadPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndMPoleSymplectic4Pass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndMPoleSymplectic4QuantPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndMPoleSymplectic4RadPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/BndStrMPoleSymplectic4Pass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/CavityPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/CorrectorPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/DeltaQPass.c
[... cut ...]
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/WiggLinearPass.c
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/ExactHamiltonianPass.cc
```

lfarv, 2021-03-04:
I get the same commands when running atmexall with Matlab R2017a on debian9, gcc 6.3 (and no problem).

Is gcc your compiler ?

Next, could you run` atmexall -v` (verbose mode)? This will get a very long print out, so I would like to see the portion corresponding to a **single** passmethod, it should be ~70 lines.
Thanks

ZeusMarti, 2021-03-05:
I'm running AT on a cluster access machine, it has installed :

`CentOS Linux release 7.9.2009`

The default GCC version is 4.8.5 (the one used earlier). In this case, the atmexall -v comands gives:

```
mex -DGLNXA64 -largeArrayDims -silent -v -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators -ldl -outdir /home/zeus/controlroom/at2_2021_test/atmat/attrack /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.c
Verbose mode is on.
... Looking for compiler 'gcc' ...
... Executing command 'which gcc' ...Yes ('/usr/bin/gcc').
... Executing command 'gcc -print-file-name=libstdc++.so' ...Yes ('/usr/lib/gcc/x86_64-redhat-linux/4.8.5/libstdc++.so').
Found installed compiler 'gcc'.
Options file details
-------------------------------------------------------------------
	Compiler location: /usr/bin/gcc
	Options file: /home/zeus/.matlab/R2017a/mex_C_glnxa64.xml
	CMDLINE2 : /usr/bin/gcc -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64 -shared  -O -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map" /tmp/mex_24496144391149596_25055/atpass.o /tmp/mex_24496144391149596_25055/c_mexapi_version.o   -ldl   -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -o /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.mexa64
	CC : /usr/bin/gcc
	DEFINES : -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE 
	MATLABMEX : -DMATLAB_MEX_FILE 
	CFLAGS : -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread
	INCLUDE : -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators  -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include"
	COPTIMFLAGS : -O -DNDEBUG
	CDEBUGFLAGS : -g
	LD : /usr/bin/gcc
	LDFLAGS : -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64
	LDTYPE : -shared 
	LINKEXPORT : -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map"
	LINKLIBS : -ldl   -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++
	LDOPTIMFLAGS : -O
	LDDEBUGFLAGS : -g
	OBJEXT : .o
	LDEXT : .mexa64
	SETENV : CC="/usr/bin/gcc"
CXX="g++"
CFLAGS="-ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
CXXFLAGS="-ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
COPTIMFLAGS="-O -DNDEBUG"
CXXOPTIMFLAGS="-O -DNDEBUG"
CDEBUGFLAGS="-g"
CXXDEBUGFLAGS="-g"
LD="/usr/bin/gcc"
LDXX="g++"
LDFLAGS="-pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64 -shared  -ldl   -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map""
LDDEBUGFLAGS="-g"
	GCC : /usr/bin/gcc
	CPPLIBS : /usr/lib/gcc/x86_64-redhat-linux/4.8.5/libstdc++.so
	MATLABROOT : /mnt/hpcsoftware/MATLAB/R2017a
	ARCH : glnxa64
	SRC : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.c;/mnt/hpcsoftware/MATLAB/R2017a/extern/version/c_mexapi_version.c
	OBJ : /tmp/mex_24496144391149596_25055/atpass.o;/tmp/mex_24496144391149596_25055/c_mexapi_version.o
	OBJS : /tmp/mex_24496144391149596_25055/atpass.o /tmp/mex_24496144391149596_25055/c_mexapi_version.o 
	SRCROOT : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass
	DEF : /tmp/mex_24496144391149596_25055/atpass.def
	EXP : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.exp
	LIB : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.lib
	EXE : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.mexa64
	ILK : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.ilk
	MANIFEST : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.mexa64.manifest
	TEMPNAME : /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass
	EXEDIR : /home/zeus/controlroom/at2_2021_test/atmat/attrack/
	EXENAME : atpass
	OPTIM : -O -DNDEBUG
	LINKOPTIM : -O
	CMDLINE1_0 : /usr/bin/gcc -c -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators  -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include" -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -O -DNDEBUG /home/zeus/controlroom/at2_2021_test/atmat/attrack/atpass.c -o /tmp/mex_24496144391149596_25055/atpass.o
	CMDLINE1_1 : /usr/bin/gcc -c -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I/home/zeus/controlroom/at2_2021_test/atmat/../atintegrators  -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include" -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -O -DNDEBUG /mnt/hpcsoftware/MATLAB/R2017a/extern/version/c_mexapi_version.c -o /tmp/mex_24496144391149596_25055/c_mexapi_version.o
	CMDLINE3_0 : rm -f /tmp/mex_24496144391149596_25055/atpass.o
	CMDLINE3_1 : rm -f /tmp/mex_24496144391149596_25055/c_mexapi_version.o
-------------------------------------------------------------------
[... 28 lines cut ...]
0xa032af crash_signal
	../../gcc/toplev.c:333
0xc38181 wi::from_mpz(tree_node const*, __mpz_struct*, bool)
	../../gcc/wide-int.cc:268
0xaff347 number_of_iterations_lt
[... cut ...]
	../../gcc/predict.c:1501
0x96aa69 tree_estimate_probability()
	../../gcc/predict.c:2349
0x96c1e2 execute
	../../gcc/predict.c:3043
Por favor, envíe un informe completo de errores,
con el código preprocesado si es apropiado.
Please include the complete backtrace with any bug report.
Véase <http://gcc.gnu.org/bugs.html> para instrucciones.
Error in atmexall (line 59)
mex(mexargs{:});
```

By the way, when I change to Matlab to R2019b, the module automatically changes gcc to 9.3.0 wich works fine.

lfarv, 2021-03-08:
@ZeusMarti : the listing you sent shows the compilation of `atpass.c`, which is not an integrator. Integrators start at the 5th `mex` statement. They are quite different: usually, all mex files have a single entry point, called `mexFunction`. Integrators need a second one, called `trackFunction`. This is what appears to be missing when you run AT. To declare this additional entry point, `atmexall` needs to generate a link option which depends on the OS, on the Matlab version and on the compiler. This what I am looking for in your output.

For the compiler error, I don't think there is anything we can do…

ZeusMarti, 2021-03-09:
OK, I see, here is the atmexall -v result for the first integrator (AperturePass):

```
mex VERSIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -v -outdir /home/zeus/controlroom/at2_2021_test/atintegrators /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.c
Verbose mode is on.
... Looking for compiler 'gcc' ...
... Executing command 'which gcc' ...Yes ('/usr/bin/gcc').
... Executing command 'gcc -print-file-name=libstdc++.so' ...Yes ('/usr/lib/gcc/x86_64-redhat-linux/4.8.5/libstdc++.so').
Found installed compiler 'gcc'.
Options file details
-------------------------------------------------------------------
	Compiler location: /usr/bin/gcc
	Options file: /home/zeus/.matlab/R2017a/mex_C_glnxa64.xml
	CMDLINE2 : /usr/bin/gcc -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64 -shared  -O -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map" /tmp/mex_25491784581031940_76018/AperturePass.o /tmp/mex_25491784581031940_76018/c_mexapi_version.o   -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -o /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.mexa64
	CC : /usr/bin/gcc
	DEFINES : -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE 
	MATLABMEX : -DMATLAB_MEX_FILE 
	CFLAGS : -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread
	INCLUDE : -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include"
	COPTIMFLAGS : -O -DNDEBUG
	CDEBUGFLAGS : -g
	LD : /usr/bin/gcc
	LDFLAGS : -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64
	LDTYPE : -shared 
	LINKEXPORT : -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map"
	LINKLIBS : -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++
	LDOPTIMFLAGS : -O
	LDDEBUGFLAGS : -g
	OBJEXT : .o
	LDEXT : .mexa64
	SETENV : CC="/usr/bin/gcc"
CXX="g++"
CFLAGS="-ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
CXXFLAGS="-ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
COPTIMFLAGS="-O -DNDEBUG"
CXXOPTIMFLAGS="-O -DNDEBUG"
CDEBUGFLAGS="-g"
CXXDEBUGFLAGS="-g"
LD="/usr/bin/gcc"
LDXX="g++"
LDFLAGS="-pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64 -shared  -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map""
LDDEBUGFLAGS="-g"
	GCC : /usr/bin/gcc
	CPPLIBS : /usr/lib/gcc/x86_64-redhat-linux/4.8.5/libstdc++.so
	MATLABROOT : /mnt/hpcsoftware/MATLAB/R2017a
	ARCH : glnxa64
	SRC : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.c;/mnt/hpcsoftware/MATLAB/R2017a/extern/version/c_mexapi_version.c
	OBJ : /tmp/mex_25491784581031940_76018/AperturePass.o;/tmp/mex_25491784581031940_76018/c_mexapi_version.o
	OBJS : /tmp/mex_25491784581031940_76018/AperturePass.o /tmp/mex_25491784581031940_76018/c_mexapi_version.o 
	SRCROOT : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass
	DEF : /tmp/mex_25491784581031940_76018/AperturePass.def
	EXP : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.exp
	LIB : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.lib
	EXE : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.mexa64
	ILK : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.ilk
	MANIFEST : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.mexa64.manifest
	TEMPNAME : /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass
	EXEDIR : /home/zeus/controlroom/at2_2021_test/atintegrators/
	EXENAME : AperturePass
	FUNCTIONMAP : /home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.map
	VERSIONMAP : /home/zeus/controlroom/at2_2021_test/atintegrators/mexFunctionGLNX86.mapext
	OPTIM : -O -DNDEBUG
	LINKOPTIM : -O
	CMDLINE1_0 : /usr/bin/gcc -c -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include" -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -O -DNDEBUG /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.c -o /tmp/mex_25491784581031940_76018/AperturePass.o
	CMDLINE1_1 : /usr/bin/gcc -c -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"/mnt/hpcsoftware/MATLAB/R2017a/extern/include" -I"/mnt/hpcsoftware/MATLAB/R2017a/simulink/include" -ansi -fexceptions -fPIC -fno-omit-frame-pointer -pthread -O -DNDEBUG /mnt/hpcsoftware/MATLAB/R2017a/extern/version/c_mexapi_version.c -o /tmp/mex_25491784581031940_76018/c_mexapi_version.o
	CMDLINE3_0 : rm -f /tmp/mex_25491784581031940_76018/AperturePass.o
	CMDLINE3_1 : rm -f /tmp/mex_25491784581031940_76018/c_mexapi_version.o
-------------------------------------------------------------------
[... cut ...]
/usr/bin/gcc -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64 -shared  -O -Wl,--version-script,"/mnt/hpcsoftware/MATLAB/R2017a/extern/lib/glnxa64/mexFunction.map" /tmp/mex_25491784581031940_76018/AperturePass.o /tmp/mex_25491784581031940_76018/c_mexapi_version.o   -L"/mnt/hpcsoftware/MATLAB/R2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -o /home/zeus/controlroom/at2_2021_test/atintegrators/AperturePass.mexa64
rm -f /tmp/mex_25491784581031940_76018/AperturePass.o
rm -f /tmp/mex_25491784581031940_76018/c_mexapi_version.o

```

lfarv, 2021-03-13:
The "link' line (end but 2) is indeed wrong, the --version-script points to the standard export file instead of AT's one. What I get here with Matlab 2017a is the following:
```
mex VERSIONMAP="/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.mapext" FUNCTIONMAP="/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.map" -DGLNXA64 -largeArrayDims -silent -v -outdir /home/esrf/laurent/dev/libraries/at/atintegrators /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.c
Verbose mode is on.
... Looking for compiler 'gcc' ...
... Executing command 'which gcc' ...Yes ('/usr/bin/gcc').
... Executing command 'gcc -print-file-name=libstdc++.so' ...Yes ('/usr/lib/gcc/x86_64-linux-gnu/6/libstdc++.so').
Found installed compiler 'gcc'.
Options file details
-------------------------------------------------------------------
	Compiler location: /usr/bin/gcc
	Options file: /home/esrf/laurent/.matlab/R2017a/mex_C_glnxa64.xml
	CMDLINE2 : /usr/bin/gcc -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64 -shared  -O -Wl,--version-script,/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.mapext /tmp/mex_136214812775295066_32384/AperturePass.o /tmp/mex_136214812775295066_32384/c_mexapi_version.o   -L"/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -o /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.mexa64
	CC : /usr/bin/gcc
	DEFINES : -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE 
	MATLABMEX : -DMATLAB_MEX_FILE 
	CFLAGS : -fexceptions -fPIC -fno-omit-frame-pointer -pthread
	INCLUDE : -I"/mnt/multipath-shares/mcssware/com/matlab_2017a/extern/include" -I"/mnt/multipath-shares/mcssware/com/matlab_2017a/simulink/include"
	COPTIMFLAGS : -O -DNDEBUG
	CDEBUGFLAGS : -g
	LD : /usr/bin/gcc
	LDFLAGS : -pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64
	LDTYPE : -shared 
	FUNCTIONMAP : /home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.map
	VERSIONMAP : /home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.mapext
	LINKEXPORT : -Wl,--version-script,/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.map
	LINKEXPORTVER : -Wl,--version-script,/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.mapext
	LINKLIBS : -L"/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++
	LDOPTIMFLAGS : -O
	LDDEBUGFLAGS : -g
	MWCPPLIB : "/mnt/multipath-shares/mcssware/com/matlab_2017a/sys/os/glnxa64/libstdc++.so.6"
	OBJEXT : .o
	LDEXT : .mexa64
	SETENV : CC="/usr/bin/gcc"
                CXX="g++"
                CFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                CXXFLAGS="-fexceptions -fPIC -fno-omit-frame-pointer -pthread -std=c++11 -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE "
                COPTIMFLAGS="-O -DNDEBUG"
                CXXOPTIMFLAGS="-O -DNDEBUG"
                CDEBUGFLAGS="-g"
                CXXDEBUGFLAGS="-g"
                LD="/usr/bin/gcc"
                LDXX="g++"
                LDFLAGS="-pthread -Wl,--no-undefined -Wl,-rpath-link,/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64 -shared  -L"/mnt/multipath-shares/mcssware/com/matlab_2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -Wl,--version-script,/home/esrf/laurent/dev/libraries/at/atintegrators/mexFunctionGLNX86.map"
                LDDEBUGFLAGS="-g"
	GCC : /usr/bin/gcc
	CPPLIBS : /usr/lib/gcc/x86_64-linux-gnu/6/libstdc++.so
	MATLABROOT : /mnt/multipath-shares/mcssware/com/matlab_2017a
	ARCH : glnxa64
	SRC : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.c;/mnt/multipath-shares/mcssware/com/matlab_2017a/extern/version/c_mexapi_version.c
	OBJ : /tmp/mex_136214812775295066_32384/AperturePass.o;/tmp/mex_136214812775295066_32384/c_mexapi_version.o
	OBJS : /tmp/mex_136214812775295066_32384/AperturePass.o /tmp/mex_136214812775295066_32384/c_mexapi_version.o 
	SRCROOT : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass
	DEF : /tmp/mex_136214812775295066_32384/AperturePass.def
	EXP : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.exp
	LIB : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.lib
	EXE : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.mexa64
	ILK : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.ilk
	MANIFEST : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.mexa64.manifest
	TEMPNAME : /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass
	EXEDIR : /home/esrf/laurent/dev/libraries/at/atintegrators/
	EXENAME : AperturePass
	OPTIM : -O -DNDEBUG
	LINKOPTIM : -O
	CMDLINE1_0 : /usr/bin/gcc -c -DTARGET_API_VERSION=700  -DGLNXA64  -DMX_COMPAT_64  -DUSE_MEX_CMD   -D_GNU_SOURCE -DMATLAB_MEX_FILE  -I"/mnt/multipath-shares/mcssware/com/matlab_2017a/extern/include" -I"/mnt/multipath-shares/mcssware/com/matlab_2017a/simulink/include" -fexceptions -fPIC -fno-omit-frame-pointer -pthread -O -DNDEBUG /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.c -o /tmp/mex_136214812775295066_32384/AperturePass.o
[... cut ...] re/com/matlab_2017a/bin/glnxa64" -lmx -lmex -lmat -lm -lstdc++ -o /home/esrf/laurent/dev/libraries/at/atintegrators/AperturePass.mexa64
```

The 1st line, generated by `atmexall`, is identical, but there are a few differences in its processing. You are using a custom option file:

`Options file: /home/zeus/.matlab/R2017a/mex_C_glnxa64.xml`

You could try removing it (it defaults to the standard one), or regenerating it with:

`>> mex -setup`

Otherwise I don't see what is different in our setups