atcollab/at #233: pyAT compatibility with python 3.9.1

swhite2401, 2021-03-08:
Hello @lfarv @willrogers , I ahve been reported that AT fails with python 3.9.1 on Mac OS

```
ring = at.load_mat(’./hmba.mat’, mat_key=’RING’)
I get the following error: TypeError: at.lattice.elements.Element.__init__() got multiple values for keyword argument'Energy'
```

Any idea on what the problem may be?

willrogers, 2021-03-08:
I seem to remember seeing that error but that call is working for me with Python 3.9.1 on Linux.

Also the tests are running on Python 3.9 on Windows, Linux and Mac.

Is this person using the latest `master` branch?

Imabeaver, 2021-03-08:
Hello, 
I could just re-install it all from the beginning because the version I'm using was downloaded about 6 months ago and it is still working fine with the MATLAB. What is happening is that when I develop setup.py in venv and run `py.test test` I get 21 errors of functions that did not pass. As @swhite2401 wrote above, when I enter python and try to load the ring it returns the error of either having `multiple values for keyword argument'Energy' or 'Voltage'.` 
What do you think?

willrogers, 2021-03-08:
I expect that this was fixed more recently than 6 months ago, so you will need a newer version than the one you are using.

Imabeaver, 2021-03-08:
I pulled the master branch and still get completely same results. Here's a part of the error:

```
(venv) milicarakic@Milicas-MacBook-Pro pyat % python3
Python 3.9.1 (v3.9.1:1e5d33e9b9, Dec  7 2020, 12:10:52) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import at
>>> from at.load import load_mat
>>> ring = load_mat('test_matlab/S28F.mat', mat_key='LOW_EMIT_RING_INJ')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/load/matfile.py", line 116, in load_mat
    return Lattice(ringparam_filter, matfile_generator, abspath(filename),
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/lattice/lattice_object.py", line 117, in __init__
    super(Lattice, self).__init__(elems)
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/lattice/lattice_object.py", line 609, in params_filter
    for idx, elem in enumerate(elem_iterator(params, *args)):
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/load/matfile.py", line 69, in ringparam_filter
    for elem in elem_iterator(params, *args):
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/load/matfile.py", line 56, in matfile_generator
    yield element_from_dict(kwargs, index=index, check=check, quiet=quiet)
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/load/utils.py", line 238, in element_from_dict
    element = cls(*(arg for arg in elem_args if arg is not None), **elem_dict)
  File "/Users/milicarakic/Docs/MATLAB/at/pyat/at/lattice/elements.py", line 514, in __init__
    super(RFCavity, self).__init__(family_name, length,
TypeError: at.lattice.elements.LongElement.__init__() got multiple values for keyword argument 'Voltage'
```

willrogers, 2021-03-08:
I can't find the file `S28F.mat`. Can you try with `hmba.mat`?

Imabeaver, 2021-03-08:
In the end I rebooted my system, now both `S28F.mat` and `hmba.mat` work. Thank you for your help!

lfarv, 2021-03-08:
I tried here with MacOS 11.2.2, Python 3.9.2: no problem !
This looks very strange, in particular if it disappears with a reboot… Let us know if the problem reappears, with the maximum of information on the circumstances.

closed 2021-08-10 by swhite2401