atcollab/at #235: Bug in ringpara for momentum compaction factor

carmignani, 2021-03-15:
Hello,

ringpara does not compute correctly the momentum compaction factor for the ESRF storage ring lattice. It could be due to the presence of the dipole-quadrupoles, because the computation is correct for the old ESRF lattice, but I'm not sure.
ringpara computes the momentum compaction factor with this formula:
`alphac = I1/circ;`
An easy fix would be to use the function mcf, unless we want to keep using only formulas inside ringpara.

The momentum compaction of ringpara is used in the function atBunchLength.

lfarv, 2021-03-15:
@carmignani: `alphac` is indeed `I1 /circ`. So we can suspect a bug in the computation of `I1`.
I cannot reproduce the problem with the `'hmba'` test lattice, which has dipole-quadrupoles.
So can you make sure you are using the up-to-date version of AT and post the lattice which causes the problem? Thanks!

carmignani, 2021-03-16:
Hello @lfarv,
I don't have the hmba test lattice, but I did some more tests and it looks like the problem is not at the dipole-quadrupoles, but at the RF cavity pass methods.
I tried with a few lattices I had and in all cases the momentum compaction factor computed with ringpara is different than the one of the mcf function if the cavity pass method is not IdentityPass. With atradoff the two results are the same.

swhite2401, 2021-03-16:
Hell @lfarv @carmignani , I believe this relates to a more generic problem in AT: optics calculations are not valid for 6D motion. Same goes for mcf that uses findorbit4.
I am working on this problem and will have a solution to propose most likely by the end of the week. Meanwhile I suggest (as advised by @lfarv in the past) to turn OFF radiations and cavities for this kind of calculations.

lfarv, 2021-03-16:
@swhite2401, @carmignani : 
There is no "generic problem in AT" ! The momentum compaction factor is a **lattice** ("geometric") property, which CANNOT be computed in 6D ! The dispersion is the trajectory of a particle with CONSTANT off-momentum, how can you compute it if you encounter a cavity or radiation? What can be done in other codes is to **ignore** cavities when computing beta functions and dispersion. AT makes the things clear: you have to do it explicitly, which is for me the best approach.
So don't loose time in trying to find a solution to a non-existing problem: I'll immediately add the test which will prevent from producing wrong results. You have been warned !

@carmignani: the hmba test lattice is part of the distribution, with a few other ones, in the `at/pyat/test_matlab` folder.

swhite2401, 2021-03-16:
@lfarv I do not agree. There are many applications where you would like to compute optics in the presence of radiations, for instance to understand the impact of the closed orbit. There are solutions for that that can be implemented and are implemented in other codes. These are all conventions and definitions I do not see any problems deriving the a "dispersion-like" quantity in the presence of cavities or radiation as long as you know what you are doing.

swhite2401, 2021-03-16:
By the way looking into this I have found a problem in the pyAT radiate flag that is set to True when cavity_pass!=IdentityPass and 'RadPass' not found, this prevents using find_sync_orbit after a command ring.radiation_off(cavity_pass='CavityPass')

carmignani, 2021-03-16:
For the momentum compaction factor problem, I think we should run atradoff inside ringpara to put IdentityPass in the cavity pass methods. Then about 6D optics I let you continue the discussion!

There are other lattice files in at/machine_data. Should't we have only one directory with lattices?

lfarv, 2021-03-16:
@swhite2401 : Of course the optics with radiation has to be studied! AT has also the formalism for 4D (Sagan, Rubin) and 6D (Ohmi et al. and A-matrix (missing reference)). The problem is that these are not as intuitive as beta-functions… 

What I am saying is that  6D motion cannot be represented by the canonical "beta", "dispersion"... functions. These are defined for 2D motion and have clear mathematical definitions and properties (alpha=-beta'/2, gamma+K*beta = constant, eta''+k*eta=1/rho, phi=integral(ds/beta), to mention a few). All this dis not true any more with *-like functions.

So as you said, introducing a new formalism can be done only with a clear definition of the new parameters (their meaning and properties), and based on solid references.

lfarv, 2021-03-16:
@carmignani : I take note for the lattice repository. Concerning an embedded "atradoff" with ringpara, if we were to do it, it should be the other way round: at 1st glance, there are only `find_orbit6`, `findm66` and `ohmienvelope` available for 6D motion, so it would be easier to turn it on there… Everything else needs radiation off!

lfarv, 2021-03-16:
@swhite2401 : `find_sync_orbit` really needs cavities off: it computes a closed orbit with a given path lengthening. And the path lengthening is defined for a constant energy deviation (we are back to the momentum compaction factor) !

swhite2401, 2021-03-16:
Ok then we agree! This is exactly what I am working on, for pyAT at least.

I did not know AT had things already implemented, which function is doing it? I did not see anything like this in pyAT.

The approach I am taking is to extract the symplectic part of the m66 matrix to derive the optics functions, this in principle should allow to remove the "damping" part but keep all the rest and include the effect of the closed orbit.

Concerning the momentum compaction factor, you are perfectly right, we may force radiation_off in get_mcf(), this value is clearly not defined for variable energy along the ring and it could make sense to automatized the switch off.

Sorry I got confused with sync_orbit...or course you are right

TeresiaOlsson, 2021-03-16:
It's great that you are working on this. I have had problems getting the momentum compaction to agree between AT and Tracy for the Diamond-I lattice because the mcf function doesn't work well when having dispersion at the observation point due to the use of findorbit4.

I think Tracy also uses the approach to extract parameters from the m66 matrix. If I have understood it correctly, Tracy gets the momentum compaction from the matrix without the need for an equivalent of the mcf function.

closed 2025-07-23 by swhite2401