atcollab/at #248: set_cavity_phase() fails for multiple accelerating RF systems

swhite2401, 2021-04-13:
@lfarv, I ran into problems using a lattice with 2 accelerating RF systems with different frequencies, set_cavity_phase() could not be used.

I would like to propose the following to overcome this issue:

```
    if refpts is None:
        cavities = [elem for elem in ring if isinstance(elem, RFCavity)]
    else:
        cavities = ring[refpts]
    rfv = ring.periodicity*sum(elem.Voltage for elem in cavities)
    freq = numpy.array([elem.Frequency for elem in cavities])
    print("\nThis function modifies the time reference\n"
          "This should be avoided, you have been warned!\n",
          file=sys.stderr)
    u0 = get_energy_loss(ring, method=method)
    timelag = clight / (2*pi*freq) * numpy.arcsin(u0/rfv)
    for elem, tl in zip(cavities, timelag):
        elem.TimeLag = tl
```

It is then up to the user to exclude non-accelerating RF cavities using the `refpts` argument.

swhite2401, 2021-04-13:
Other minor issue, in get_energy_loss, shouldn't we take PolynomB[0] into account for dipoles when compute the integral?

```
        def dipole_i2(dipole):
            return (dipole.BendingAngle - 
                    dipole.PolynomB[0] * dipole.Length) ** 2 / dipole.Length
```

lfarv, 2021-04-13:
@swhite2401: Your solution (exactly as proposed) is already implemented in an upcoming pull request concerning global RF cavity control !
However in case of good weather conditions, you have to expect some delay… Maybe to-night.

lfarv, 2021-04-14:
About the 3 last lines of your proposal: the frequency of the involved cavities must be the same: otherwise, the sum of voltages makes no sense. So `timelag` is a scalar and must not be zipped with `cavities`. The upcoming revised version raises an exception if the frequencies are not equal.

lfarv, 2021-04-14:
About energy loss: `PolynomB[0] `is associated with orbit corrections. The computation of I2 with the integrals takes only bending magnets into account: no quadrupoles, multipoles, correctors,… I would state that this gives the "energy loss for a perfect machine". The computation with tracking may take into account more effects, though I am not sure `PolynomB[0]` is used: for thin correctors, it would give an infinite energy loss, which does not seem to appear.

swhite2401, 2021-04-14:
In this case how do you handle multiple accelerating RF systems with different frequencies?

swhite2401, 2021-04-14:
By the way did you write the A-mat module? I am getting very nice results except that I get a sign errors w.r.t theory and the phase advance calculation as a pi uncertainty...I am starting to think there is a problem in the definition of the A-mat

lfarv, 2021-04-14:
@swhite2401:
#### RF system:
I see two ways to tune complex systems:
1. Take one system ("main" RF) as the time/phase reference, with `TimeLag`=0, and adjust the other ones in phase and voltage with respect to the main one. All this using `set_cavity` with different `refpts`, each referring to cavities with the same `Frequency` and `TimeLag`. The beam will go where it finds to adequate voltage.
2. for a simple harmonic system: use `set_cavity_phase` for the main RF, and keep `TimeLag==0` for the other ones so that they do not contribute to the accelerating voltage of the reference particle. Then personally, I would shift everything again in time to bring back the main RF as the origin, but you may disagree (this means giving to harmonic systems the opposite of the `TimeLag` computed by `set_cavity_phase` for the main RF)…

#### energy loss calculation with I2:
The contribution of a bending magnet to I2 is **nominal_bending_angle/radius**. Any additional angle due to PB[0] is orbit distortion, you should not include it without the other contributions to orbit distortion (correctors, misaligned quads). As all other radiation integrals, this computation is valid for a perfect machine, at a constant energy (radiation must be **off**, I would not try with a "tapered" machine! ). Then the reference particle does not radiate in multipoles: it's on axis. Adding closed orbit effects looks difficult because then you need to include everything… Considering the bending angles and radii of correctors, compared to the 2*pi and magnetic field of main magnets, it should be negligible. I am more concerned with the effect of energy variation along the ring which is not included (energy^4 effect).
On the other hand, the computation with tracking should include everything, at least for elements with a "RadPass" method. And obviously the thin magnets do not have it (it would give infinite numbers…).

#### A-mat
The original Matlab version is from @bnash. I just translated it for python. I remember that there are different choices to be done there: choice between conjugate complex vectors (this may cause the phase advance problem), and choice to associate each vector pair with a plane. I modified this one to ensure that no motion can be associated with 2 different planes. However this becomes arbitrary for very large coupling and it may cause jumps between 2 locations. Then it may be better to talk of motion 1, 2, 3 rather than x, z, s. Concerning A-mat computation, I have nothing else than the papers from Boaz, I still have to go through them.

bnash, 2021-04-14:
Hi Laurent and Simon, 
Yes, I wrote the [amat function](https://github.com/atcollab/at/blob/master/atmat/atphysics/LinearOptics/amat.m) function
though I think Laurent also contributed.
I checked that it transformed 6x6 transfer matrices T into 3 2x2 rotation matrices R as expected. i.e. 
A^{-1}*T*A = R
The other thing to check is that the order of x,y,z comes out right, which it seemed to in the cases I checked.
@swhite2401 , are you saying that the angles comes out negative of what they should be?
I thought that should be taken care of correctly by putting the positive mode eigenvectors before the negative mode
eigenvectors. This does have the result that A should be symplectic, which can be verified.

Sorry, haven't thought about this in awhile, but glad to follow up with you both.

swhite2401, 2021-04-15:
@lfarv 

**RF systems**

1- Maybe I am missing something but I don't think this answers my question for multiple accelerating systems with different frequencies, set_cavty_phase() need the total accelerating voltage otherwise arcsin(U0/rfv), even worse if U0>rfv, for a subset of cavities for instance, this returns nan. set_cavity_phase() is therefore unusable in these conditions.

-> potential workarounds, allow for the user to provide the total accelerating voltage:
```
set_cavity_phase(rfv=None):
    if rfv is None:
        rfv = ring.get_rf_voltage(refpts=refpts)
```

2-Ok for this

**I2**

I am presently facing a blocking point where the tracking fails without cavities because Eloss/turn is too strong, the only possible approach is therefore analytic, however omitting k0 gives wrongs results. Integrating PB[0] in the calculation will give more accurate result so what is the problem? I am not sure I understand why we have to be so restrictive. Again if you do not like it we may add a flag to use PB[0] or not or even a separate function but this functionality is needed.

**Amat**

Hi @bnash ! I'dd be glad to exchange a bit more in case you have time to follow-up on this. To give you more details: for one-turn map analysis everythin seems to be working fine, however propagating the Amat to get optics and phase adavances at different points in the machine, the angle sometimes switch by Pi for not apparent reason.
The propagation and reconstruction of the Amat from transfer matrix involves quite a lot of dot products and inversion, I am wondering whether this could be a numerical artefact...

I may provide a piece of code with the tests I have done so far in case you would like to take a look at it, I do not want to put this in a branch for now as it is not advanced enough.

lfarv, 2021-04-15:
@swhite2401:

#### RF systems
1. Forget `set_cavity_phase` ! It's there because some like it, I don't. I'm talking of `set_cavity` (the one added in #249). It lets you tune each system in frequency, voltage and phase. Sorry, I'm working on it and forgot it was not in the main branch.

#### I2
I need more details on the problematic ring!
For sure, pb[0] must not be included because it does not contribute to the bending angle of the dipole. This is the basis of BndMPoleSymplectic4pass: the bending angle must be entirely defined by 'BendingAngle'. This is how `BndMPoleSymplectic4pass` differs from `StrMPpoleSymplectic4Pass`. Otherwise it gives wrong results. If you need PB[0] to close the ring, there is a bug in the lattice definition. And `BndMPoleSymplectic4Pass` will give wrong results. Or the existing integrators are not able to represent your lattice, which may happen.
You say the integrals give wrong results: it's proportional to energy^4 * sum(theta/ro), with a constant energy along the ring. It could be replaced with sum(energy^4*theta/ro) with a varying energy, derived from the previous losses. Could this help
You are exploring new regions…

swhite2401, 2021-04-15:
I do like `set_cavity_phase()` and many others do too! I understand your point and arguments but these are really personal tastes. In some cases it is very useful and should be compatible with complex RF systems, if there is a need from users this function should exist.

I will send you the mat file in question offline with some explanations.

lfarv, 2021-04-15:
For `set_cavity_phase`, I agree to provide it, but again, I do not know how to do for complex RF system: `asin(U0/VRF)` only works for a sinusoidal total voltage. Otherwise you have to find the solution of
`sum(v_i * sin(2*pi*f_i + phi_i)) = U0` which is not trivial. Introducing `rfv` as an argument does not help, you need to enter all voltages, frequencies and phases. The easiest way is the solution 1 above, followed by tracking to determine the synchronous phase,  and then shifting all the TimeLags by  -phase_advance. You let the beam solve the equation.
And there may be several stable solutions!
Any other idea?

swhite2401, 2021-04-15:
It is a tricky issue and you are right rfv as input is not satisfactory, it is too approximate. Tracking could be a good solution to explore.

swhite2401, 2021-04-15:
Btw, feel free to merge your developments they are improvements! We may tackle the complex RF systems issues separately

swhite2401, 2021-04-15:
Just found another side effect after merging the master into the dev branch, the automated initial guess for `find_orbit6` also fails for complex RF system...we should set it to np.zeros(6) if len(frev)>1

lfarv, 2021-04-15:
Right. But it's much worse than that: orbit6 does not work at all:

> find_orbit6 finds the closed orbit in the full 6-D phase space
>     by numerically solving  for a fixed point of the one turn
>     map M calculated with lattice_pass
> 
>     (X, PX, Y, PY, DP, CT2 ) = M (X, PX, Y, PY, DP, CT1)
> 
>     with constraint  CT2 - CT1 = C*HarmNumber(1/Frf - 1/Frf0)

To solve this equation, it takes Frf and HarmNumber from the **1st cavity!** So the system which is solved to get the next iteration is wrong. Since it is an iterative search, it may still converge to a solution. But maybe not. The good point is that if it converges, it's a correct solution.

I'll at least secure the initial guess, but for the rest, I do not know how to do.

swhite2401, 2021-04-15:
To be honest I don't either... I think the first thing is to secure the present implementation and then we have to think a bit more concerning the more complex situations.

bnash, 2021-04-15:
Hi @swhite2401 and @lfarv, I created [a new issue](https://github.com/atcollab/at/issues/251) to discuss the amat questions Simon raised, which is pretty separate from the main topic of this issue.

closed 2021-08-10 by swhite2401