atcollab/at #250: Bug in atradon

carmignani, 2021-04-15:
Hello,
I updated the repository today and some functions don't work anymore. I think the problem is in atradon, which calls endsWith instead of endswith, and endsWith is in atoctave directory and fails with matlab.
This breaks atx, which calls atradon, and some other functions calling atradon.

https://github.com/atcollab/at/blob/dc956ff21e1aaba2ec741ac381748e5d3f70d843/atmat/atphysics/Radiation/atradon.m#L78

lfarv, 2021-04-15:
Very strange: `endsWith` appeared in Matlab 2016b, which is the oldest Matlab version AT is compatible with. What is the Matlab version you are using ?

carmignani, 2021-04-15:
Ok, sorry, endsWith is the correct spelling, but there is another endsWith in atoctave that must be not in the matlab path, otherwise it is used instead of the one of matlab.

carmignani, 2021-04-15:
So this is not a bug, but still this means that we can't just add all the directories of the repository to the matlab path, which is fine, but a bit annoying.

lfarv, 2021-04-15:
`atoctave` implements functions which are not yet in Octave. Indeed, it should **not** be in Matlab path. Only `atmat` and `atintegrators`. Do you confirm the problem, or can we close the issue ?

lfarv, 2021-04-15:
I looked at the` Install.md`, which indicates the installation procedure: the method calling `atpath` is correct, the "manual" method mentions `atmat` and `atintegrators`, which is correct, but misses `machine_data`. I'll correct that !
But be careful, including other things in the path (python stuff or tests for instance) may cause problems.

carmignani, 2021-04-15:
Yes, fine, we can close. I modified my startup.
I saw the endswith by searching in GitHub, but that was the python function and I got confused.

closed 2021-04-15 by carmignani