atcollab/at #251: 6D matrix normalization, amat and phase advance

bnash, 2021-04-15:
@swhite2401 [has raised some questions](https://github.com/atcollab/at/issues/248#issuecomment-820170170) about how to find phase advance for the general 6D optics, and whether the [A matrix](https://github.com/atcollab/at/blob/master/atmat/atphysics/LinearOptics/amat.m) is defined correctly.

We can use this issue to discuss all of this.

bnash, 2021-04-15:
[Here](https://github.com/atcollab/at/blob/master/pyat/at/physics/amat.py) is Laurent's Python version of this function.

bnash, 2021-04-15:
See also [this 1998 reference](https://journals.aps.org/pre/abstract/10.1103/PhysRevE.58.2481) by E. Forest, "Dispersive lattice functions in a six-dimensional pseudo-harmonic-oscillator"

swhite2401, 2021-04-16:
Hi @bnash,
Very good idea, so to summarize (from the tests I have done):
-getting optics function using betoids/etoids functions works very nicely
-getting the tune based on a one-turn map also work very well
-problem comes from: getting the phase advance from point 1-> 2 I get pi "jumps"

This the little code I used for testing (M->transfer map,M0->one tuen map, for from `find_m66`)
```
def get_mus(M,M0):
    A = amat(M0)
    M2 = numpy.dot(M0,inv(M))
    A2 = amat(M2)
    A2I = inv(A2)
    R12 = numpy.squeeze(multi_dot([A2I,M,A]))
    mux = numpy.arctan2(R12[0,1]-R12[1,0],R12[0,0]+R12[1,1])
    muy = numpy.arctan2(R12[2,3]-R12[3,2],R12[2,2]+R12[3,3])
    #print(M,'\n\n',M2,'\n\n',M0,'\n\n')
    #print(A,'\n\n',A2,'\n\n',AI,'\n\n',A2I,'\n\n')
    #print(A2I-AI,'\n\n',A2-A,'\n\n',M2-M0,'\n\n')
    return numpy.array([mux,muy])
```
You can see it is relatively simple, maybe I am missing something??

Interesting fact, the pi jump depends on:
-an arbitratry factor applied on M66 before computing the eigenvectors:  _,vv=eig(m66*fact)
-the method employed for inversion: inv,pinv,solve

Which makes me think we are facing numerical issue related to rounding errors.

Any advice?

Thanks!

bnash, 2021-04-16:
Hi @swhite2401, thanks for the code. Glad to hear that A works correctly for normalizing the 1-turn map matrix!

First of all, recall that A is not uniquely defined. As Forest points out, you can multiply A by a rotation matrix and it will
still normalize M0 as expected. Let A' = Ar, where r is a rotation matrix (uncoupled). Likewise A2 that normalizes M2 can be multiplied by a different rotation matrix. Say A2' = A2 r'. If you look at your code for R12, 
`R12 = numpy.squeeze(multi_dot([A2I,M,A]))`
you can see that this would change the definition of the phase advance.
So from this perspective, the phase advance is not well defined!

On the other hand, I believe that if one normalizes the A matrix such that it has the Courant snyder form:
 Ax=     | 1/sqrt(beta_x)               0           |
           | alpha_x/sqrt(beta_x)  sqrt(beta_x) |
  A = |Ax   0|
         |0   Ay|
then, the phase advance will come out as expected. This is for x, and likewise for y. I assume it's uncoupled here.
I think that all you have to do is force the (0,1) element of the A matrix to be 0, and that's enough.
I think Forest calls this freedom to multiply A by a rotation matrix r as 'gauge freedom'.
Another way to say that "the phase advance will come out as expected" is to say that the matrices A(s) are well defined.
If you compute them by applying amat() to M(s) you get one result. If you compute amat(s0) and then transform it to another s, then you may get a different result! I think this Courant-Snyder convention ensures these two ways to calculate A(s) give the same result.

Since the 'gauge' of A in our definition will be somewhat arbitrary, depending on the algorithm for finding the eigenvectors, I think we can set it explicitly to make sure the (0,1) and (2,3) elements are 0.
Suppose that
Ax = | a  b |
        | c   d|
 We find a rotation matrix rx such that Ax rx has a 0 in the (0,1) component.
 Solving for the rotation angle phi, we get 
 phi = atan(-b/a)
 And likewise for Ay.
 
 I thought I once wrote a function to do this, (based on a question from Simone), but I can't find it now.
 
 For another discussion leading to this same conclusion, see eqn 75 in Wolski
 https://journals.aps.org/prab/abstract/10.1103/PhysRevSTAB.9.024001

I'm a little confused that the result depends on how you do the inverse. From what I'm saying here, it should depend more on how the eigenvectors are computed...

Anyway, I hope this might be the relevant issue!

swhite2401, 2021-04-17:
Hi @bnash ,
Ah this is great this looks indeed like the missing piece!
Thanks a lot, I keep you updated.

swhite2401, 2021-04-17:
@bnash , there was infact two thigs:

1- A and A2 in get_mus must have the same polarity
2- There was an uncertainty on the phase just as you suggested

The 2 following functions fix these issues an the agreement is now excellent for the uncoupled case!
Let see what happens with coupling now! Thanks a lot for the very useful suggestion!

```
def set_rmat(A):
    select = numpy.array([0,2,4],dtype=numpy.uint32)
    r3 = numpy.zeros((6,6))
    for s in select:
       t = numpy.arctan(A[s,s+1]/A[s,s])
       ci = numpy.cos(t)
       si = numpy.sin(t)
       r = numpy.array([[ci, -si], [si, ci]])
       r3[s:s+2,s:s+2] = r
    return numpy.dot(A,r3)

def equalize_rmat(A1, A2):
    signs = numpy.array(sign(A1+1e-24)*
                        sign(A2+1e-24))
    R = numpy.identity(6)*signs
    return numpy.dot(A2,R)
```

lfarv, 2021-04-17:
Hi all,
Thanks @bnash for this new reference, everything is there go on. A few remarks:
- I find the output in terms of Bk (formula 58) much better that in terms of betaoids: it puts the results in a more "usual" way, with the betas in position [0,0], [2,2], alphas in [1,0], [3,2] etc. And no etaoid is necessary for the dispersion!
- it covers also the treatment made in `atx` by processing the matched distribution
- the so-called "standardization" (line 77) is very simple.
- 
Also, I intend to make this available not only in 6x6, but also in 4x4 for lattices without RF cavity where the 3rd rotation is missing.

The funny thing is that `atx` which already analyses the distribution matrix as in III.B already gave all that (but kept them hidden…)

bnash, 2021-04-19:
Looks like nice progress @swhite2401 and @lfarv !
This issue of phase advance is subtle and I always felt like I only 3/4 understood it!
I'll keep following the discussion and let you know if I have anything else constructive to contribute.
Regards,
Boaz

swhite2401, 2021-04-19:
@bnash, indeed it looks like most critical points have been sorted out, I am not sure I fully understood everything either... but at least it works as expected.

@lfarv, I agree with you it looks like these are slightly simpler formulas and I fully agree that we should not restrict this to 6x6 matrices.
How would you like to proceed? I have a prototype that runs and seems to be working for pyAT, however it is way below standards for a pull request so it is probably best to start from scratch now that the method is established, also nothing was done for matlab. Do you have anything in mind?

lfarv, 2021-04-19:
@swhite2401: A first draft should be ready by the end of the day. Both python and Matlab. Notations based on Wolski's paper. At the moment, it's still missing the "transfer line" option, but for non-coupled machines, results are strictly identical to `linopt`.

swhite2401, 2021-04-19:
@lfarv, ah you are very efficient! Very nice! Then I am looking forward to see your implementation!

lfarv, 2021-04-19:
1st draft now available for tests in #252

closed 2021-08-10 by swhite2401