atcollab/at #266: Tutorials and manuals for pyAT

elafmusa, 2021-06-25:
Hi,

For Matlab AT there are some tutorials and manuals on the web, but I could not find similar things for pyAT.
Do you have something like that?

I'm a new PhD student  starting to look into optics calculations and corrections, and it would be good if i can do some pyAT training first.

Thanks,

lfarv, 2021-06-27:
@elafmusa : you are right, pyAT is still missing documentation…
I could start looking at that, but I do not know what format to use. Any suggestion, @willrogers, @matsievskiysv ?

matsievskiysv, 2021-06-27:
Relevant [discussion](https://github.com/atcollab/at/discussions/237):
> @lcarver sent me the tutorial. It was great. I think that's exactly what people would want. The tutorial explains how to create a lattice using the elements definitions, combine it into a lattice and save as a mat file that can be imported when doing a simulation without the need for having other codes. That is what I meant by a pyAT lattice file.

lcarver, 2021-06-28:
[PyAT_Lattice.pdf](https://github.com/atcollab/at/files/6724031/PyAT_Lattice.pdf)

Hello @elafmusa, welcome to the world of AT/PyAT

I have attached a tutorial / basic command overview for PyAT, This tutorial is most definitely a WIP that I have not got around to finishing. Maybe it is a good side project for me to finish during the summer weeks...

The tutorial does not discuss the physics of PyAT/AT as the physics is the same as matlab AT, which is very well documented in various tutorials and documents that can be found online. There are a few inconsistencies in this tutorial (which is why it is a WIP) that you should be careful of:

1. I use two similar functionalities interchangeably, the following two lines perform the same action and the help reads differently from the tutorial

[ld0, tunes, chroma, ld] = at.linopt(ring, dp=0)
[ld0, tunes, chroma, ld] = ring.linopt(dp=0)

2. The lattice that I try to design in the tutorial doesn't really match very well at the end. I noticed this at the last step and haven't yet got around to fixing it. Nonetheless the lattice definition and syntax are correct so there is still a lot to be gained from taking this version.

Please do not hesitate to ask if you have questions or problems. 

Good luck!

elafmusa, 2021-06-28:
> [PyAT_Lattice.pdf](https://github.com/atcollab/at/files/6724031/PyAT_Lattice.pdf)
> 
> Hello @elafmusa, welcome to the world of AT/PyAT
> 
> I have attached a tutorial / basic command overview for PyAT, This tutorial is most definitely a WIP that I have not got around to finishing. Maybe it is a good side project for me to finish during the summer weeks...
> 
> The tutorial does not discuss the physics of PyAT/AT as the physics is the same as matlab AT, which is very well documented in various tutorials and documents that can be found online. There are a few inconsistencies in this tutorial (which is why it is a WIP) that you should be careful of:
> 
> 1. I use two similar functionalities interchangeably, the following two lines perform the same action and the help reads differently from the tutorial
> 
> [ld0, tunes, chroma, ld] = at.linopt(ring, dp=0)
> [ld0, tunes, chroma, ld] = ring.linopt(dp=0)
> 
> 1. The lattice that I try to design in the tutorial doesn't really match very well at the end. I noticed this at the last step and haven't yet got around to fixing it. Nonetheless the lattice definition and syntax are correct so there is still a lot to be gained from taking this version.
> 
> Please do not hesitate to ask if you have questions or problems.
> 
> Good luck!

Hello @lcarver ,

This is much appreciated! Thank you a lot.
I'll start looking at the tutorials and contact you if needed.

Best Regards,
Elaf

willrogers, 2021-09-22:
See also https://atcollab.github.io/at/pprimer.html

@lfarv must have quietly added this recently but it's a very good introduction. Then I will close this ticket.

closed 2021-09-22 by willrogers