atcollab/at #269: PyAT: Wrong Tau in envelope_parameters

lcarver, 2021-08-03:
Hello,
I have noticed that envelope_parameters is giving the wrong damping time. But radiation_parameters is correct.

```
ring = at.load_mat('./S28F.mat', mat_key = 'LOW_EMIT_RING_INJ')
env_r = ring.envelope_parameters()
env_r.Tau 
```
`array([-1.40685286e+05,  1.59258359e+06,  4.87636845e+08])`

```
ring.radiation_off()
env2 = ring.radiation_parameters()
env2.Tau
```
`array([0.00870282, 0.01333954, 0.00909173])`

Seems to be computed a bit differently and the error is coming from Ohmi envelope. I will explore further.

lcarver, 2021-08-03:
The problem comes from the fact that now envelope_parameters will run with radiation_off and return 0 for the emittances and the huge numbers for the damping rates. I think no 'fix' is needed, but think we should have a print warning if radiation is off when running envelope_parameters.

swhite2401, 2021-08-09:
@lcarver, the S28F lattice (at least the one I have) has cavities but radiations OFF, so no damping. 
The calculation is therefore correct (infinite damping times). If radiation flag is OFF is fails with an ATError as expected. 

I could not reproduce the case where Ohmi_Envelop forces radiations ON, in fact the decroators @check_radiation(True) is present so I don't think that is possible to run it with radiations OFF

The reason why it works in radiation_parameters is that in this case the calculation is analytical and
does not need radiations.

There is certainly an ambiguity in the naming as ring.radiation() returns True even without radiation in case an RF cavity is present...but I have not good proposal to make it more intuitive

closed 2021-08-10 by swhite2401