atcollab/at #296: Release of PyAT 0.2.0

willrogers, 2021-09-16:
I would like to use some of the newer functions in pyat, so I intend to release version 0.2.0. Does anyone have any comments before I do so?

I will try to make this compatible with Python 3.9 (but not Python 2).

Aside: I am leaving Diamond in October, so I may be making no further contributions to PyAT! @mjgaughran will pick up some of this work so you might see him ask questions or make contributions.

swhite2401, 2021-09-16:
Ok for me, although it would have been nice to merge #271 first to be consistent with the latest developments in `linopt`.
@lfarv did you take a look?

Good luck for the future and welcome to @MJGaughran!

lfarv, 2021-09-16:
I agree for #271. But can you wait one more day for another pull request related to linopt?

@willrogers: We'll miss you! Anyway good luck, and welcome @MJGaughran!

swhite2401, 2021-09-17:
I merged #271, ok for me to release the new version

lfarv, 2021-09-17:
Ok for version 0.2.0.
Concerning #290, up to now it corrects a small part of the problem, only in a PassMethod which is almost never used, so it can be merged, but it really does not matter.
@willrogers : you need to explain the procedure to create a new release in pip, and where this procedure is defined.

willrogers, 2021-09-17:
It is not straightforward to support Python 3.9 since our supported numpy version (1.14.3) does not work.

There are other mechanisms to handle this but they are not trivial, so I propose to release pyat-0.2.0 supporting 3.5 - 3.8 and consider Python 3.9 later.

lfarv, 2021-09-17:
PyAT runs smoothly on Python 3.9. Isn't it possible to define the numpy minimum version to what is necessary for Python 3.9?
Anyway, no problem to consider Python 3.9 later.

willrogers, 2021-09-19:
I've just experimented and it looks like we could support Python 3.5 to 3.9 with Numpy 1.16.6.

Just because it builds doesn't necessarily mean it will run... but I think we should be ok.

willrogers, 2021-09-20:
I ran all tests using numpy 1.16.6. There is one problem (one test fails) on MacOS for Python 3.8 (and probably 3.9) with that specific version - otherwise all tests pass. The problem goes away for newer versions of Numpy, which will usually be what is installed.

https://github.com/atcollab/at/runs/3649562724?check_suite_focus=true

Since this is a small problem with one platform with an old version of numpy that is not likely to be used, and a warning is printed, I will continue with Numpy 1.16.6.

lfarv, 2021-09-20:
I run daily on MacOS and have no problem. So go on with NumPy 1.16.6. In addition the reported problem (`init_dgelsd failed init`) is well known and is not linked to the NumPy version, but comes from a wrong installation.

willrogers, 2021-09-20:
I have released pyat 0.2.0: https://pypi.org/project/accelerator-toolbox/

I am updating the documentation for how to upload. I will also need to add some other people as maintainers for accelerator-toolbox. I will add @mjgaughran, would you like this as well @lfarv ?

willrogers, 2021-09-23:
@mjgaughran is now going to make a release 0.2.1 after #301 was fixed. Then I will close this ticket.

willrogers, 2021-09-24:
Pyat 0.2.1 released.

closed 2021-09-24 by willrogers