atcollab/at #315: pyat: pip install . does not work

willrogers, 2021-10-05:
As reported by @jbengtsson. This is something to do with the file copying process not working.

This is listed in the documentation as a way to install pyat.

```
$ pip3 install .
Processing /tmp/atbuild/at/pyat
  Installing build dependencies ... done
Requirement already satisfied: numpy>=1.16.6 in /tmp/atbuild/venv/lib/python3.7/site-packages (from accelerator-toolbox==0.2.1) (1.21.2)
Requirement already satisfied: scipy>=0.16 in /tmp/atbuild/venv/lib/python3.7/site-packages (from accelerator-toolbox==0.2.1) (1.7.1)
Installing collected packages: accelerator-toolbox
  Found existing installation: accelerator-toolbox 0.2.1
    Uninstalling accelerator-toolbox-0.2.1:
      Successfully uninstalled accelerator-toolbox-0.2.1
  Running setup.py install for accelerator-toolbox ... error
    Complete output from command /tmp/atbuild/venv/bin/dls-python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-2b9wngi0/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-xkbnkd43/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/atbuild/venv/include/site/python3.7/accelerator-toolbox:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.7
    creating build/lib.linux-x86_64-3.7/at
    copying at/__init__.py -> build/lib.linux-x86_64-3.7/at
    creating build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/lattice_object.py -> build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/cavity_access.py -> build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/elements.py -> build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/utils.py -> build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/__init__.py -> build/lib.linux-x86_64-3.7/at/lattice
    copying at/lattice/options.py -> build/lib.linux-x86_64-3.7/at/lattice
    creating build/lib.linux-x86_64-3.7/at/tracking
    copying at/tracking/track.py -> build/lib.linux-x86_64-3.7/at/tracking
    copying at/tracking/particles.py -> build/lib.linux-x86_64-3.7/at/tracking
    copying at/tracking/patpass.py -> build/lib.linux-x86_64-3.7/at/tracking
    copying at/tracking/__init__.py -> build/lib.linux-x86_64-3.7/at/tracking
    creating build/lib.linux-x86_64-3.7/at/plot
    copying at/plot/generic.py -> build/lib.linux-x86_64-3.7/at/plot
    copying at/plot/__init__.py -> build/lib.linux-x86_64-3.7/at/plot
    copying at/plot/synopt.py -> build/lib.linux-x86_64-3.7/at/plot
    copying at/plot/specific.py -> build/lib.linux-x86_64-3.7/at/plot
    creating build/lib.linux-x86_64-3.7/at/integrators
    copying at/integrators/pyDriftPass.py -> build/lib.linux-x86_64-3.7/at/integrators
    copying at/integrators/__init__.py -> build/lib.linux-x86_64-3.7/at/integrators
    copying at/integrators/pyIdentityPass.py -> build/lib.linux-x86_64-3.7/at/integrators
    creating build/lib.linux-x86_64-3.7/at/load
    copying at/load/allfiles.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/matfile.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/tracy.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/elegant.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/utils.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/__init__.py -> build/lib.linux-x86_64-3.7/at/load
    copying at/load/reprfile.py -> build/lib.linux-x86_64-3.7/at/load
    creating build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/linear.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/ring_parameters.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/radiation.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/harmonic_analysis.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/nonlinear.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/energy_loss.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/__init__.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/fastring.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/amat.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/matrix.py -> build/lib.linux-x86_64-3.7/at/physics
    copying at/physics/orbit.py -> build/lib.linux-x86_64-3.7/at/physics
    creating build/lib.linux-x86_64-3.7/at/matching
    copying at/matching/globalfit.py -> build/lib.linux-x86_64-3.7/at/matching
    copying at/matching/matching.py -> build/lib.linux-x86_64-3.7/at/matching
    copying at/matching/__init__.py -> build/lib.linux-x86_64-3.7/at/matching
    running build_ext
    building 'at.tracking.atpass' extension
    creating build/temp.linux-x86_64-3.7
    creating build/temp.linux-x86_64-3.7/tmp
    creating build/temp.linux-x86_64-3.7/tmp/pip-req-build-2b9wngi0
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DPYAT -I/tmp/pip-build-env-fjhvllkh/lib/python3.7/site-packages/numpy/core/include -I/tmp/pip-req-build-2b9wngi0/integrator-src -I/tmp/atmat/atphysics/Radiation -I/tmp/atbuild/venv/include -I/dls_sw/prod/tools/RHEL7-x86_64/Python3/3-7-2dls1/prefix/include/python3.7m -c /tmp/pip-req-build-2b9wngi0/at.c -o build/temp.linux-x86_64-3.7/tmp/pip-req-build-2b9wngi0/at.o -Wno-unused-function
    /tmp/pip-req-build-2b9wngi0/at.c:12:21: fatal error: attypes.h: No such file or directory
     #include "attypes.h"
                         ^
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
  Rolling back uninstall of accelerator-toolbox
Command "/tmp/atbuild/venv/bin/dls-python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-req-build-2b9wngi0/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-xkbnkd43/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/atbuild/venv/include/site/python3.7/accelerator-toolbox" failed with error code 1 in /tmp/pip-req-build-2b9wngi0/
You are using pip version 18.1, however version 21.2.4 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

willrogers, 2021-10-05:
I think @MJGaughran should have a look at this.

jbengtsson, 2021-10-05:
Roger that. (received & understood)
Thanks.

lfarv, 2021-10-11:
1. The failure depends on what happened in the `integrator-src` directory before. After cleaning it completely, I get the same missing file: `attypes.h`,

2. pip install -e . works as expected,

3. the most recent `pip` prints this:
   ```
   DEPRECATION: A future pip version will change local packages to be built in-place without first copying
   to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages
   with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at
   https://github.com/pypa/pip/issues/7555.
   ```
   And `pip install --use-feature=in-tree-build .` works.
   So it seems that `pip install` will work in the next `pip` version.

MJGaughran, 2021-10-12:
The alternative would be to refer directly to `pip install -e .`. It means you no longer need to care about the `requirements.txt` just to test the module. `pip` also just has a better interface. 

We actually have 3 different files with dependencies right now: `requirements.txt`, `setup.py` and `pyproject.toml`. There's probably a better way of organising these.

lfarv, 2021-10-12:
I think that editable install should be limited to development. For a normal user, the standard installation in site-packages is safer. 

Concerning the organisation of requirements, I have a problem:
why does `pyproject.toml `requires `numpy==1.16.6` instead of `numpy>=1.16.6` ? On my machine, installing `numpy@1.16.6 ` under python 3.9 fails: no wheel available, and build NumPy from source fails. So I cannot use pip with python 3.9, with or without -e … And again. I have to come back to setup.py.
Changing to `numpy>=1.16.6`, pip takes the latest available wheel and works.

Another reason to refer to python `setup.py` install in the documentation.

lfarv, 2021-10-15:
@MJGaughran : this looks very good!

closed 2022-11-18 by swhite2401