atcollab/at #325: Integral tolerances in Touschek Piwinski lifetime in matlab

carmignani, 2021-11-09:
I did some Touschek lifetime simulations varying different parameters for the ESRF storage ring and I noticed that with the default parameters often we get not precise results, at least for very large vertical emittances.
 
I attach here a plot of the Touschek lifetime computed with TouschekPiwinskiLifeTime as a function of vertical emittance.
The blue is using the method "quad", that is not recommended by matlab, but it gives the good results. 
The other ones are instead using the "integral" method, but changing the relative and absolute tolerances. The red curve is with the default values of tolerances of "integral", which is what we get by running TouschekPiwinskiLifeTime with default parameters.

I think the best is to allow the user to give the absolute and relative tolerances as input (now you have to modify the function to change the tolerances) and to give a smaller number as default. It seems like 1e-16 and 1e-12 are small enough for the case I studied.

[image]

simoneliuzzo, 2021-11-09:
I see only benefit in this change.

swhite2401, 2021-11-10:
I agree. Why is quad not recommended? Looking at these curves I would just set it as default...
Btw, I did an implementation in python using `integrate.quad` that I still need to push at some point that seems to work very well

carmignani, 2021-11-10:
I don't know exactly why quad is not recommended: this is what they write in the documentation.
[image]

lfarv, 2021-11-14:
See #326 for an answer to this!

lfarv, 2021-11-17:
Solved in #326

closed 2021-11-17 by lfarv