atcollab/at #328: rin format changed in python passmethods

swhite2401, 2021-11-16:
@lfarv in PR #321 I think you have changed the format of the input of the python passmethods going form `(6*n,)` to `(6,n)`  by using `rin` instead of drin. The concerned source in `at.c` is:

```
            if (*pyintegrator) {
                #old implementation
                if (!*kwargs) *kwargs = Buildkwargs(*element);
                *kwargs = pyIntegratorPass(drin, *pyintegrator, *kwargs, num_particles);
                if (!*kwargs) return print_error(elem_index, rout);       /* trackFunction failed */
                #new implementation
                PyObject *res = PyObject_CallFunctionObjArgs(*pyintegrator, rin, *element, NULL);
                if (!res) return print_error(elem_index, rout);       /* trackFunction failed */
                Py_DECREF(res);
```

A user complained his code is broken... however, my feeling is that the present implementation is more inline with the python convention for particles that is `(6,n)`. I tempted to keep this new format and ask him to modify his code accordingly.
We will need to update the example passmethods and tests that should include several particles in order to avoid this kind of issue to re-appear.

What do you think?

lfarv, 2021-11-17:
The new implementation sends to the python integrator the original python object (`rin`), as it was sent to `at_atpass`. For me it's better than rebuilding a new python object from the double array (`drin`), and giving it different dimensions. So I agree with you to keep the present version and consider the previous implementation as a bug.

OK for improving the tests!

swhite2401, 2021-11-17:
Ok then it is done, I will add a new PR with modified examples and improved tests

closed 2021-11-17 by lfarv