atcollab/at #335: Wrong definition of projected emittances?

swhite2401, 2021-12-09:
Hello,
It was pointed out to me by I. Asparuhov and A. Franchi, that what is defined as the projected emittances in AT is calculated as:
`eps = sqrt(|sigma|)`

while the proper calculation for projected emittance should be:
`eps_{i,j} = sqrt(|inv(inv(sigma)[i,j:i,j])|)`

I have looked in the litterature and I think they are right, what is called projected emittance in AT is generally defined as the rms emittance.
If confirmed we need at least to correct the help  with the proper definition and we may add an additional field with 3 projected emittance (xxp,yyp,sdp).

What do you think?

lfarv, 2021-12-09:
Sorry for I. Asparuhov and A. Franchi, but AT is right.

What is called "projected emittance" is the projection (mathematical definition) of the 6D emittance on the selected axes (x-x' or z-z'). So it includes all energies and chromatic effects, and of course varies along the ring because of the dispersion and possible x/z coupling.

On the other hand, what you get by taking a subset of the inverse matrix is a "cut" of the 6D emittance for &delta;=0, ct=0, z=0, z'=0. I would rather call this one "monochromatic emittance" or "betatron emittance" (usually invariant along the circumference).

The term rms emittance ia ambiguous and may be applied to any of those.

More generally, if take a subset of the sigma matrix, you get a projection on the axes that you keep. Look at sigma(0,0): it's the square of the overall beam size, including energy spread and dispersion.

If you take a subset of the inverse matrix, you get a cut for all the ignored axis at zero.

swhite2401, 2021-12-10:
Ok nothing to be done then, I close it.

closed 2021-12-10 by swhite2401