atcollab/at #338: Relativistic  particles

swhite2401, 2021-12-16:
Presently only relativistic particle are allowed with `rest_energy=0`
This causes a problem in all calculations using `ring.gamma` which is coming out as `inf`

For example `ring_parameters.radiation_parameters` returns many `nans` and `inf` which is
clearly an issue.
We probably have to find a better way to handle this, I would suggest to move this to `ring.beta`
that could have a boolean keyword ultra-relativistic for example, default particle could then be electrons

lfarv, 2021-12-16:
Initially, all calculations assumed electrons, except tracking which assumes &beta;=1. The present version brings consistence between tracking and processing. But as long as the non-relativistic tracking is not validated, AT is consequently blocked on "relativistic particle".
To solve the problem, I propose to switch back to a default of "electron" (temporarily non-modifiable), which will turn all the outputs back to what they were before, but to force the tracking to &beta;=1. This way:
- Everything will be back to what it was before,
- Using 1 instead of the electron &beta; should not make any visible difference.

swhite2401, 2021-12-17:
Ok for me

swhite2401, 2021-12-17:
However @simoneliuzzo expressed some interest in modeling ultra-relativitic hadrons, to allow this we just have to set beta=1 regardless of the particle mass.
We may issue a warning is the "real" beta differs too much from 1.

lfarv, 2021-12-21:
Corrected in #339